ARM-software/CMSIS_5 #2: RTOS: There are no benefits in having negative priorities

ilg-ul, 2016-02-27:
I could not detect any benefits on using negative values for priorities; unsigned values seem a better match for them, if for no other reasons, at least for matching what debuggers usually display.

As it is now, a priority of -1 is shown in a memory map as 0xFF, which is confusing.

For the future version of CMSIS RTOS, my suggestion is to use unsigned numbers.

Reserve **0** for **undefined** values (to catch uninitialised variables), **1** for the **IDLE** thread, and go up to the max range value, then count down and reserve **(max-1)** to mark the **error** case and also reserve **(max-2)**  for the system deferred **ISR** thread (for a complete priority range see issue #1).

Regards,

Liviu

ReinhardKeil, 2016-02-29:
Hi Liviu,

thanks for the feedback. We are right now discussing it.

Reinhard

RobertRostohar, 2016-03-09:
There are no real benefits of having negative values (osPriorityNormal was initially 0 which lead to negative values for lower priorities).

Priorities will be changed to positive numbers with other modifications (see issues #1).

RobertRostohar, 2016-04-08:
Priority values have been changed to positive values.

f595671e8228efec6acdd5681d817f0b28604ad8

closed 2016-04-08 by RobertRostohar