ARM-software/CMSIS_5 #13: SVDConv version 3.2.17 does not like 0xFFFFFFFF in <enumeratedValue>

David-Garcia-Polo, 2016-03-09:
If an SVD file defines a register with a field of 32 bits and one of the <enumeratedValue> has <value> 0xFFFFFFFF (perfectly valid 32-bit number, and quite popular), SVDConv fails with error M369. The very similar 0xFFFFFFFE does not fail.

Please see attached SVD files for example. Rename to .svd since Github does not support .svd extension.

[M369_error.svd.txt](https://github.com/ARM-software/CMSIS_5/files/165391/M369_error.svd.txt)
[M369_no_error.svd.txt](https://github.com/ARM-software/CMSIS_5/files/165392/M369_no_error.svd.txt)

thorstendb-ARM, 2016-03-09:
fixed in SVDConv V3.2.25

ReinhardKeil, 2016-06-30:
David, the new version is available here:
https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/Utilities

Do you have a chance to test/review the new version and close the related issues?

David-Garcia-Polo, 2016-06-30:
yes. will test today.

ReinhardKeil, 2016-09-05:
Closed now as there is no further activity

closed 2016-09-05 by ReinhardKeil