ARM-software/CMSIS_5 #15: SVDConv version 3.2.17 cannot define to <enumeratedValue> with the same <value>

David-Garcia-Polo, 2016-03-09:
SVDConv generates warning M333 when two <enumeratedValue> in the same register have the same <value>. It is very useful to substitute deprecated names, but cannot currently be done. I do not see any reason in the documentation why this would not be allowed.

Can M333 be changed into an info instead? If not, can the disabling of warnings change the return code as well, not just the log?

See attached files.
[M333_warning.svd.txt](https://github.com/ARM-software/CMSIS_5/files/165476/M333_warning.svd.txt)
[M333_no_warning.svd.txt](https://github.com/ARM-software/CMSIS_5/files/165477/M333_no_warning.svd.txt)

thorstendb-ARM, 2016-03-09:
SVDConv V3.2.25 does not count suppressed warnings, this is also reflected by the return value.
We will discuss M333.

ReinhardKeil, 2016-09-05:
Closed now as there is no further activity

closed 2016-09-05 by ReinhardKeil