ARM-software/CMSIS_5 #33: Compiler warning in CMSIS c startup files

kjelleandersen, 2016-04-18:
Compiling the following files emit warnings when compiled with gcc and the -Wpedantic flag:

warning: ISO C forbids conversion of object pointer to function pointer type [-Wpedantic]
   (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
   ^

Files affected:
4.5.0/Device/ARM/ARMCM0/Source/GCC/startup_ARMCM0.c:136:          (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMCM0plus/Source/GCC/startup_ARMCM0plus.c:136:  (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMCM3/Source/GCC/startup_ARMCM3.c:140:          (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMCM4/Source/GCC/startup_ARMCM4.c:140:          (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMCM7/Source/GCC/startup_ARMCM7.c:140:          (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMSC000/Source/GCC/startup_ARMSC000.c:136:      (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
4.5.0/Device/ARM/ARMSC300/Source/GCC/startup_ARMSC300.c:140:      (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */

GuentherMartin, 2016-04-26:
Hello,
a possible solution is to change the type of the external __STACKTop.

    //extern uint32_t __StackTop;
    extern void __StackTop(void); /* Declared as function to avoid compiler warning. */

    const pFunc __Vectors[] __attribute__ ((section(".vectors"))) = {
      /* Cortex-M4 Exceptions Handler */
    //  (pFunc)&__StackTop,                       /*      Initial Stack Pointer     */
      __StackTop,                               /*      Initial Stack Pointer     */
      Reset_Handler,                            /*      Reset Handler             */

Let me know your opinion.

kjelleandersen, 2016-04-26:
Why not be pragmatic and just be a little more explisit when casting ?

  (pFunc)((uint32_t)&__StackTop),           /*      Initial Stack Pointer     */

I also think that values originating from linker cmd file should always be ints or uints (not sure here)
It feels a bit scary to handle stacktop as a function, C does not assume that a func pointer even has the same number of bits as a void pointer.
And could bit 0 (thumb/arm mode) come into play and add to the confusion ?

Best regards,
Kjell Eirik Andersen
Staff Software Engineer
IoT MCU & Wireless
_________________________________________
Silicon Labs
Mobile: +47 484 90 451 | Direct: + 47 23 00 98 00 www.silabs.com<http://www.silabs.com>

GuentherMartin, 2016-04-26:
You are right!
This solution is pragmatic and clearer.

The other solution looked also scary to me but it avoided the warning and worked in simulation.

ReinhardKeil, 2016-09-05:
Closing this now as there is no further activity

closed 2016-09-05 by ReinhardKeil