ARM-software/CMSIS_5 #46: Inconsequent NVIC_ functions

franzhollerer-Artesyn, 2016-04-22:
There is the following code in core_cm0.h:

```c
__STATIC_INLINE void NVIC_SetPendingIRQ(IRQn_Type IRQn)
{
  NVIC->ISPR[0U] = (uint32_t)(1UL << (((uint32_t)(int32_t)IRQn) & 0x1FUL));
}

__STATIC_INLINE void NVIC_ClearPendingIRQ(IRQn_Type IRQn)
{
  NVIC->ICPR[0U] = (uint32_t)(1UL << (((uint32_t)(int32_t)IRQn) & 0x1FUL));
}

__STATIC_INLINE void NVIC_SetPriority(IRQn_Type IRQn, uint32_t priority)
{
  if ((int32_t)(IRQn) < 0)
  {
    SCB->SHP[_SHP_IDX(IRQn)] = ((uint32_t)(SCB->SHP[_SHP_IDX(IRQn)] & ~(0xFFUL << _BIT_SHIFT(IRQn))) |
       (((priority << (8U - __NVIC_PRIO_BITS)) & (uint32_t)0xFFUL) << _BIT_SHIFT(IRQn)));
  }
  else
  {
    NVIC->IP[_IP_IDX(IRQn)]  = ((uint32_t)(NVIC->IP[_IP_IDX(IRQn)]  & ~(0xFFUL << _BIT_SHIFT(IRQn))) |
       (((priority << (8U - __NVIC_PRIO_BITS)) & (uint32_t)0xFFUL) << _BIT_SHIFT(IRQn)));
  }
}

__STATIC_INLINE uint32_t NVIC_GetPriority(IRQn_Type IRQn)
{

  if ((int32_t)(IRQn) < 0)
  {
    return((uint32_t)(((SCB->SHP[_SHP_IDX(IRQn)] >> _BIT_SHIFT(IRQn) ) & (uint32_t)0xFFUL) >> (8U - __NVIC_PRIO_BITS)));
  }
  else
  {
    return((uint32_t)(((NVIC->IP[ _IP_IDX(IRQn)] >> _BIT_SHIFT(IRQn) ) & (uint32_t)0xFFUL) >> (8U - __NVIC_PRIO_BITS)));
  }
}
```

What is interesting is that the functions NVIC_SetPriority() and NVIC_GetPriority() check the passed IRQn number. If IRQn is negative they access the System control block (SCB), otherwise the Nested vectored interrupt controller (NVIC).

With this, these functions can be used for system exceptions _AND_ NVIC interrupts.

This is different from the other functions. For NVIC_SetPendingIRQ() and NVIC_ClearPendingIRQ() the passed IRQn number must be positive. They don't support system exceptions, even it would be nice if NVIC_SetPendingIRQ() could be used to trigger a PendSV  exception.

Therefore, I have the following objections:

- If the functions support negative IRQn numbers, the prefix NVIC_ is misleading. SetIRQPriority(), GetIRQPriority() or similar may be a better name.
- If NVIC_SetPriority(), NVIC_GetPriority() support system exceptions (i.e. a negative IRQn number), the other NVIC_ functions should do so as well.
- The doxygen comment before each function should state if negative IRQn numbers are allowed or not.

GuentherMartin, 2016-05-09:
We kept the current functionality but updated the documentation and also the functions according negative IRQn numbers.

ReinhardKeil, 2016-07-12:
closing as there is no further reply

closed 2016-07-12 by ReinhardKeil