ARM-software/CMSIS_5 #48: Core-Only SVD files

hackrid, 2016-04-22:
I have found these files which do not contain the register description for the core peripherals like NVIC and SCB:
https://github.com/ARM-software/CMSIS_5/tree/develop/Device/ARM/SVD

are these popualted anytime soon?

jkrech, 2016-04-22:
The Core Peripherals are well understood and displayed in the available toolchains already today, therefore the priority to invest here is currently very low. Nevertheless we are always open for contributions.

hackrid, 2016-04-22:
thanks for your response.
maybe the efforts are as low as the priority :)
do these files exist already?

jkrech, 2016-04-22:
Nope.

hackrid, 2016-04-22:
regarding the state in "available toolchains":
if you compare these two files, you will find that one has the SCB peripheral in it an the other not.

These files were delivered with Keil MDK.

SFD\Freescale\Kinetis\MK40D7.sfr vs. SFD\ST\STM32F4xx\STM32F401x.SFR

I think there is still a lot to be fixed...

jkrech, 2016-04-22:
Note that the corresponding SVD files are provided by the Silicon Vendors.
Which version of Keil MDK are you looking at ? From the folders it looks like Version 4.x.
Device Support in Version 5.x is distributed as CMSIS-Packs. The device family packs also contain the latest SVD files.

ilg-ul, 2016-04-22:
@hackrid, allow me to translate the ARM statement "The Core Peripherals are well understood and displayed in the available toolchains already today":

> We, ARM, have the definitions of the core peripherals, and use them in our development tools, but we will not release them separately because we want you to use our tools. 

normally the place for the ARM core peripherals is not in the vendor files, because there are hundreds of them and it would be highly redundant.

the description for the core peripherals should be provided by ARM, in the ARM core packs, and the development tools should be ready to use them to complement the vendor definitions, in other words to first load the vendor definitions, and then check if there are more core peripherals not yet loaded, and add them.

at least this will be the strategy that I plan to use in [GNU ARM Eclipse](http://gnuarmeclipse.github.io), as soon as the core definitions will be available.

ReinhardKeil, 2017-02-27:
It should be noted that we made progress here: the Keil.V2M-MPS2_CMx_BSP.1.6.0.pack contains SVD files that describe also the CORE peripherals.  Can you check if this would fullfil your requirements?

JonatanAntoni, 2017-03-13:
@hackrid may I ask you to take a look to the packs Reinhard mentioned? Please close this issue if you do not have further questions.

JonatanAntoni, 2017-03-21:
Closed due to inactivity.

closed 2017-03-21 by JonatanAntoni