ARM-software/CMSIS_5 #54: Is the variable SystemCoreClock used by any debugger?

franzhollerer-Artesyn, 2016-05-02:
The CMSIS standard specifies that the SystemCoreClock variable can be used by the debugger to query the frequency of the debug timer or to configure the trace clock speed. Therefore, the compiler must be configured to avoid removing this variable in case the application program is not using it.

[image]

I wonder if there is any debugger using this variable or plans to use it in future. For the Keil toolchain, it seems that SystemCoreClock is not used.

[image]

Furthermore, it seems that there is no plan to use it in future (see feature request, case id 658081).

I have the impression that the SystemCoreClock is a very interesting and promising idea, but it is not used as originally intended.

Thus, if no debugger is using it, I suggest to weaken the requirements on its implementation. For applications which don't need to support low power modes, and therefore don't change the system clock, a definition as macro somewhere at a header file would be more efficient. This would allow the user to let the compiler perform every clock related calculation at compile time.

JonatanAntoni, 2017-03-24:
Currently we are not aware of any debuggers or third party tools relying on `SystemCoreClock`. The intend of this variable has not changed. But it depends on your individual setup and requirements. If you are sure about the usage of this variable in your environment, it is up to you to remove it.

Unfortunately you have to change the system header of your device which is not a user file:

`extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */`

must be changed to

`#define SystemCoreClock SYSTEM_CORE_CLOCK  /*!< System Clock Frequency (Core Clock) */`

I discussed this issue here and we might enhance the system header like

`#ifdef SYSTEM_CORE_CLOCK  
#define SystemCoreClock SYSTEM_CORE_CLOCK
#else
extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */
#endif`

This solution would give you the choice. But it needs to be incorporated by all the device family pack suppliers.

JonatanAntoni, 2017-03-24:
Short note to myself.

Changes needed:
- system device header
- system device source (user template containing SystemCoreClockUpdate)
- vendor specific clock configuration component

Taking a simply Blinky example with ARM Compiler 6 and opt level -O1 shows:
- 252 bytes less ROM usage (2%)
- 4 bytes less RAM usage

Using opt level -Os (balanced) shows:
- 244 bytes less ROM usage (2.1%)
- 4 bytes less RAM usage

Using opt level -Oz (image size) shows:
- 212 bytes less ROM usage (1.9%)
- 4 bytes less RAM usage

franzhollerer-Artesyn, 2017-03-27:
Thanks for considering these changes. A few things I want to add for completeness. First off all I must admit that some use cases will benefit from having the system clock defined as macro, others not. If you have to change the clock speed during runtime, SystemCoreClock implemented as variable might be the better choice.

My applications typically run with a fixed speed. I usually have a config file, e.g. project_config.h with compile time configurations for the project. This also defines a macro for the clock speed.

project_config.h:

```c
#define CONFIG_CPU_SYSCLK_HZ 16000000
#define CONFIG_CPU_PCLK1_PRESCALER 1
```

At the moment I initialize the SystemCoreClock variable with CONFIG_CPU_SYSCLK_HZ for the case a debugger needs this information.

system_[device].c:

```
#include "project_config.h"

uint32_t SystemCoreClock __attribute__((used)) = CONFIG_CPU_SYSCLK_HZ;
```

My own code uses the macro, not the variable. The benefit is that if the compiler sees the constant at compile time, and therefore you can use floating point terms to calculate prescalers for timer, UART etc. and finally cast it to an integer type before using it. The armcc is smart enough to perform the floating point operations at compile time. With this you don't need to care about integer over- or underflows, and the code keeps runtime optimized.

For example to calculate the clock divider for a UART:

```c
#define BAUD_RATE 115200
#define PCLK1_HZ (CONFIG_CPU_SYSCLK_HZ / CONFIG_CPU_PCLK1_PRESCALER)

#define USARTDIV \
    MATH_ROUND_TO(unsigned, (double) PCLK1_HZ / BAUD_RATE)
```

With MATH_ROUND_TO() being a support macro defined somewhere else:

```c
/**
 * Round a floating point value to an integer type at compile time.
 *
 * This macro is intended apply rounding at compile time. If you need
 * to round a floating-point value to an integer type at runtime, please
 * use round(), roundf() or similar rounding functions provided by
 * the C standard library.
 *
 * @param[in] type The target data type.
 * @param[in] val  The floating point value to round to the given type.
 *
 * @returns Rounded value cast to the desired type.
 *
 * Example
 *
 * @code
 * int foo = MATH_ROUND_TO(int, 1.75);
 * @endcode
 *
 * Will evaluate to
 *
 * @code
 * int foo = 2;
 * @endcode
 *
 * at compile time, at least if compiler optimization be turned on.
 *
 * @note
 * The compiler is not required to evaluate a floating point expression
 * at compile time, but all compilers we have seen so far are doing
 * this. On doubt please check the compiler output.
 */
#define MATH_ROUND_TO(type, val) \
    ((type)( ((val) < 0) ? ((val) - 0.5) : ((val) + 0.5) ))
```

Being said, having SystemCoreClock as macro instead of a variable gives the following benefits:
- prescalers / dividers for timer, UART, etc. can be calculated at compile time
- you can use floating point expressions, which are resolved at compile time (provided that a) you know what you do and b) the compiler is smart enough).

ReinhardKeil, 2018-05-28:
To my knowledge, debuggers are not yet using this value.  However the value is used to configure the RTOS system tick.

ReinhardKeil, 2018-05-29:
Closing it now as strictly speaking its not related to CMSIS but to debug implementations.

closed 2018-05-29 by ReinhardKeil