ARM-software/CMSIS_5 #57: CMSIS Drivers: Instance parameters needed

0x8000-0000, 2016-05-15:
[Note: I have opened the same issue against CMSIS4 https://github.com/ARM-software/CMSIS/issues/89  It might well be that the CMSIS4 API is frozen, but maybe there's a chance this API be improved for CMSIS5]

I find the API for the CMSIS drivers, specifically the USART and SPI confusing. They seem to assume there is only one instance of USART or SPI peripheral in use, or that different peripherals will have different function implementations.

For instance, this is the ARM_USART_Send signature:

  int32_t ARM_USART_Send (const void *data, uint32_t num);

If my application / board uses both USART1 and USART2, how do I differentiate between the two?

I understand that I could write ARM_USART_Send_using_USART1 function and ARM_USART_Send_using_USART2 function, and populate each in the respective ARM_DRIVER_USART but that seems quite non-intuitive, especially when on a piece of silicon all similar functional blocks have similar if not identical capabilities, down to the register interface. Ignoring that USART1 runs on a faster bus than USART2 and might need slightly different initialization sequences, the read and the write are identical (save for the base address of the register set).

ilg-ul, 2016-05-15:
> I have opened the same issue against CMSIS4 but it might well be that the CMSIS4 API is frozen, but maybe there's a chance this API be improved for CMSIS5.

the whole point of having multiple GitHub repositories for the same project is meaningless; I strongly advised against it, but apparently ARM did not get the point. the explanation that this is required by the license change is laughable (at best).

> They seem to assume there is only one instance of USART or SPI peripheral in use, 

yes, this is a possible explanation.

my analysis is a bit deeper, to me it seems they have no experience with C++ and object oriented design at all. once you start thinking with objects, it becomes very clear that everywhere you need to pass a pointer to the object instance. as such, the reentrant callbacks (see #86) is one of the places where CMSIS Drivers definitely failed.

0x8000-0000, 2016-05-15:
Liviu:

> to me it seems they have no experience with C++ and object oriented design at all

If you look at the STM32 libraries, Freescale SDK, Energy Micro libraries, they all have functions with a base/handle/"this" pointer as the first argument. So, if one follows those examples or wants to implement the CMSIS driver by delegating to the manufacturer's library, the mismatch would be quite apparent.

ilg-ul, 2016-05-15:
well, your observation is right, but I'm not sure it is more flattering for ARM. on the contrary, I would say.

ReinhardKeil, 2016-06-30:
Before we continue discussion you should review how we solve such things in practice.

- http://www.keil.com/pack/doc/CMSIS/Driver/html/_theory_operation.html#AccessStruct
Read topic driver instances

- And an example implementation (in this case for identical SPI channels) is here:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Pack/Example/CMSIS_Driver/SSP_LPC18xx.c

Overall the complexity and overhead is minor.

*Why did we no use a pointer to a device instance:*
For identical peripherals it is very compelling to use a device instance pointer, but in practice we have different technology that provides same functionality. For SPI we have in the same device sometimes: simple SPI hardware, multi-purpose peripherals that can be USART, SPI, I2C, and MCI hardware that could be also used as SPI channel.  Managing all that via a device instance pointer becomes a nightmare.

If you expose device instance pointers to middleware for storage it would become difficult to support: external SPI memory, internal on-chip Flash, external Flash in same application.  At some point all needs to be managed in a driver implementation file that deals with instances. Using the driver struct approach allows you to use different implementation files for each of the technologies and all you need is to expose the struct of each implementation.

ReinhardKeil, 2016-07-12:
We are closing issues that are marked DONE after 30 days when there is not further reply

ReinhardKeil, 2016-09-05:
Closed now as there is no further activity

closed 2016-09-05 by ReinhardKeil

ilg-ul, 2016-09-05:
You closed this, but you did not even bother to understand the issue.