ARM-software/CMSIS_5 #59: CMSIS-RTOS: Ability to add a name to a Queue/Semaphore

hackrid, 2016-05-24:
FreeRTOS offers this in the incarnation of :http://www.freertos.org/vQueueAddToRegistry.html

this is an important feature of RTOS aware debugging

ilg-ul, 2016-05-24:
agree, and should apply to all objects, not only to queues/semaphores.

CMSIS++ supported named objects from the very beginning, and recently the name was moved from the relative obscure location inside the object attributes to a more visible location, in the object constructors.

actually all object constructors are doubled, one without name and one with name (on the first position).

as for C, the CMSIS++ RTOS C API also allows defining names to all objects.

for more detail please see:

https://github.com/micro-os-plus/cmsis-plus/blob/xpack/include/cmsis-plus/rtos/os-c-api.h

ReinhardKeil, 2016-06-30:
We are going to publish CMSIS-RTOS2 in a few days.  This feature (and much more) is added there.

ilg-ul, 2016-10-12:
according to the current API, names can be set via attributes, but the entire use case is too tedious,
see also #66.

object names are a **very** important aspect of RTOS aware debugging; apart from RTX, which is in its own world, most other RTOSes use names for objects (FreeRTOS, µC/OS, SMX, ThreadX, etc, not to mention CMSIS++).

JonatanAntoni, 2017-02-27:
Is this issue resolved in the meantime, due to all RTOSv2 objects can have a name attribute now? Or do you have any further suggestions regarding names for RTOS-aware debugging?

ilg-ul, 2017-02-27:
objects can have names, but setting them via attributes is equivalent to saying that these names are seldom used.

in practice, it is recommended to set them always, at least in Debug mode. this means RTOSv2 objects **always** are created using attributes. this requires several lines of code for each object, which is very tedious.

in another issue (#66) I suggested to make these names first citizens, and pass them in the object creation call.

ReinhardKeil, 2017-02-28:
It should be noted that debugging is not the main driver for "named" objects.  Good debuggers show objects in context, even when there is no name assigned. In reality user defined ASCII strings are critical as there is no guarantee that a name is unique.  The motivation to add names is cross platform references in multi-core implementations.

ilg-ul, 2017-02-28:
> ... Good debuggers show objects in context, even when there is no name assigned.

we already had this discussion before. debuggers are able to show symbols only for statically allocated objects; dynamically allocated objects, either in a pool or in the heap do not have much context to show, so at run time the user provided string is a real help.

of course it is the user responsibility to assign these strings uniquely, but failing to do so is not critical, normally these strings are not actively used in deployed applications. if the application does rely on them, the RTOS should check the uniqueness at object creation time, but this is a different issue.

JonatanAntoni, 2017-03-24:
I think we should limit this discussion to the issues title. We have enriched the object attributes with a name field. The name field is only intended for identifying the objects. We do not have plans to check uniqueness of names at all.

Coming back to the concerns about having the name attribute not available as a first class citizens. We aim to keep all the API functions as short as possible, i.e. using four parameters at maximum. Refer to [Clean Code by Robert C. Martin](https://books.google.co.uk/books?id=_i6bDeoCQzsC&pg=PR14&lpg=PR14&dq=%22F1:+Too+Many+Arguments%22&source=bl&ots=eobKAk9e21&sig=nA_6CeNZoZieQHaisVm7QyCyCXg&hl=en&sa=X&ved=0ahUKEwj68LSnj-_SAhWIDMAKHfbRAgQQ6AEIIjAB#v=onepage&q=%22F1%3A%20Too%20Many%20Arguments%22&f=false), he states:

> F1: Too Many Arguments
> Functions should have a small number of arguments. No argument is best, followed by one, two, and three. More than three is very questionable and should be avoided with prejudice.

Thus we decided to put all attributes to the (optional) attribute structure, instead of discussion first vs. second class citizen for each new option. This gives us the following benefits:
* The attribute struct can be made const and placed to ROM.
* Up to four parameters can be transferred via registers, instead of using the stack.
* Using C99 struct initialization allows us to enhance the struct without breaking user code.

Please be so kind and close this issue if our naming solution is good enough to you for the moment.

ilg-ul, 2017-03-24:
what you are basically doing is to discourage adding names to objects.

JonatanAntoni, 2017-03-24:
I am sorry for making an discouraged impression to you. If there is nothing else I can do about this issue, please let us close it.

ilg-ul, 2017-03-24:
I'm not the author of this issue.

JonatanAntoni, 2017-04-03:
I am closing this issue due to no further comment on the provided solution.

closed 2017-04-03 by JonatanAntoni