ARM-software/CMSIS_5 #60: SVDConv do not generate SysTick_IRQn.

David-Garcia-Polo, 2016-06-27:
some of our designs have no SysTick timers, but I have not found a way to describe in the SVD file that no sysTick is available. Header files generated by SVDConv.exe contain SysTick functionality. when our customers use it, it does not work (it is not there) and contact support. 

We would like a way to have SVDConv not generate SysTick_IRQn.

ilg-ul, 2016-06-27:
> some of our designs have no SysTick timers

can you be more specific which one?

David-Garcia-Polo, 2016-06-27:
Nordic Semiconductor nRF51822 device or nRF51422 device have no SysTick, neither the standard ARM nor any vendor SysTick.

ilg-ul, 2016-06-27:
> nRF51822 or nRF51422

ok, thank you. 

just wanted to be sure I'll be able to avoid them completely.

David-Garcia-Polo, 2016-06-27:
dont be so mean :)

ilg-ul, 2016-06-27:
FYI, I've just completed the reference implementation of the [CMSIS++ RTOS](http://micro-os-plus.github.io/cmsis-plus/), and all time related operations, be it a simple osDelay(), thread preemption, blocking calls timeouts, thread usage statistics, etc are all based on SysTick, as they should be.

designing a Cortex-M device without a SysTick is like putting a large banner "don't try to use me with an RTOS, I'm not suitable for this".

salkinium, 2016-06-27:
> designing a Cortex-M device without a SysTick is like putting a large banner "don't try to use me with an RTOS, I'm not suitable for this".

SysTick is usually connected to the CPU clock, meaning if you want to sleep the CPU, there goes your SysTick.
Replacing the SysTick with a low-power timer is a good idea for when you want to keep an accurate time base in low-power designs, but need to wake up periodically (ie. lower than typical RTC time periods).

There is more than one solution to design an RTOS, you should see this as an opportunity, rather than a limitation.

ilg-ul, 2016-06-27:
> There is more than one solution to design an RTOS

the CMSIS++ RTOS design is quite elaborated, it specifies three clocks: the system clock (that counts ticks when non sleeping), the real time clock (that can be used while sleeping) and the high resolution clock (that counts clocks when accurate measurements are required). this does not mean that further improvements are no longer possible, but within limits. ARM invented SysTick exactly to help simplify software, and providing it in real devices is kind of a curtesy to developers. You might save a quarter of a microamp by designing proprietary hardware, but this will also require proprietary software, which adds to the total cost.

salkinium, 2016-06-27:
> this does not mean that further improvements are no longer possible, but within limits

Indeed, we're barely beginning to understand energy efficient software.
Why would tick based scheduling be the most energy efficient way of scheduling? ;-P

>  but this will also require proprietary software, which adds to the total cost.

You will need new software to take advantage of energy-efficient hardware anyway. Just throwing an RTOS at it won't cut it. It's more of a brute force approach to the problem, which is quite a bit more complicated in detail. Just think of subtree clock gating and how that could be integrated into an RTOS. The "custom" timer is the least of your problems. 

Don't loose track of what's ahead. This is the golden time of µC OS design. We are seeing whole new approaches to adding light-weight concurrency that are not thread based (coroutines, fibres), and integrating energy awareness into them. It's going to be EPIC.

`</rave>`

David-Garcia-Polo, 2016-06-28:
as salkinium just defended, our design has a propietary timer that allows the core to sleep while counting. Therefore we do not have a systick.

ilg-ul, 2016-06-28:
> a proprietary timer ...

the CMSIS++ / µOS++ project is highly portable, and if someone will be interested to port it to Nordic devices, the solution is a copy/paste of the generic Cortex-M port-specific files (3 headers and one cpp) and an update to use the proprietary timer; I guess this shouldn't be a major problem, if Nordic wants to do it I see no reasons for not including this specific port in the distribution.

I'm not sure if this can be also applied to other systems, especially commercial systems, delivered as binary libraries.

salkinium, 2016-06-28:
> I'm not sure if this can be also applied to other systems, especially commercial systems, delivered as binary libraries.

Like the Bluetooth Softdevices from Nordic? 😜

ReinhardKeil, 2016-06-30:
Thanks for pointing to that weakness. We will review and take action.

Note: For CMSIS-RTOS RTX a user timer can be defined, in cases where no SYSTICK timer is available.

jkrech, 2016-11-25:
Please see the SVD documentation about the <cpu> section here:
http://arm-software.github.io/CMSIS_5/SVD/html/elem_cpu.html

In the SVD file you need to set the tag 
<device>
 ...
  <cpu>
     ...
     <vendorSystickConfig>true</vendorSystick/Config>
     ...
  </cpu>
  ...
</device>

Please let me know if this fixes your problem.

jkrech, 2016-11-28:
SVDConv Version 3.2.56 does not generate SysTick_IRQn anymore when <vendorSystickConfig> true.

jkrech, 2016-11-28:
Sorry for closing.  David, please close if the problem is resolved to your satisfaction.

David-Garcia-Polo, 2016-12-07:
Where can I ger v3.2.56? the one checked in is v3.2.53

jkrech, 2016-12-07:
Version 3.2.56 should be the latest on the develop branch of the CMSIS_5 repo. What am I missing ?

David-Garcia-Polo, 2016-12-07:
It is me, I was taking master branch.

closed 2016-12-07 by David-Garcia-Polo