ARM-software/CMSIS_5 #61: Inclusion hell within RTX

0xc0170, 2016-06-30:
Headers do not include required headers to fetch declarations, types. Just random inclusion in some cases, no protect guards for multiple inclusion. State it in the header file the intention at least, that it's private, and it's included only in one code file because of some reasons.
``RTX_CM_lib.h``.is a nice example of it.

Files referenced:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS/RTX/SRC/rt_HAL_CM.h
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS/RTX/SRC/RTX_Config.h
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS/RTX/INC/RTX_CM_lib.h

I am trying to change some configuration, as I am working on modified RTX kernel because of some reasons, everytime I have to edit kernel, I come across this issue.

ReinhardKeil, 2016-11-15:
RTX5 is cleaner in this aspect. I hope you can cope with the shortcomings of RTX4.

0xc0170, 2017-01-03:
Thanks for having a look at this issue. I had a quick look at RTX5 include headers, looks good to me. I'll close this as resolved (as I understand RTX4 stays as it is).

closed 2017-01-03 by 0xc0170