ARM-software/CMSIS_5 #62: Private stack size in t he P_TCB - u16 -> u32?

0xc0170, 2016-07-06:
The original issue reported on mbed: https://github.com/mbedmicro/mbed/pull/826 , in the ``P_TCB`` structure there's ``priv_stack`` member that is declared as ``u16``. Please consider having private stack as 32bit variable.

Reference : https://github.com/ARM-software/CMSIS_5/blob/4868c88b763f9923330359cec1e522555f093777/CMSIS/RTOS/RTX/SRC/rt_TypeDef.h#L73

We found out  that in DS-5 for RTOS for Cortex-A includes this change (u32 for priv_stack) but only if ``__LARGE_PRIV_STACK`` is defined. Do you got any more info for this condition?

ReinhardKeil, 2016-11-17:
RTX5 is using 32-bit values

ReinhardKeil, 2017-02-13:
RTX 5 fixes this issue.  Now it is also available as IAR port

closed 2017-02-13 by ReinhardKeil