ARM-software/CMSIS_5 #63: Library issues with SVDConv.linux

jmcfarlan, 2016-07-07:
When trying to use the SVDConv.linux in our environment, we get the following:

./SVDConv.linux: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by ./SVDConv.linux)
./SVDConv.linux: /tools/cadence/incisiv/14.20.006/tools/lib/64bit/libstdc++.so.6: version `GLIBCXX_3.4.15' not found (required by ./SVDConv.linux)

It seems that our environment in stuck at RedHat 6.x which limits us to GLIBC_2.12.  Is there any way to modify the path that this utility searches for to find the libraries?

jkrech, 2016-07-08:
Could you please download the SVDConv Linux build using GCC 4.4.4 from here and let me know whether this would work for you?

www.keil-compiler.de/Transfer/Preview/SVDConv_gcc-4.4.4.zip

Redirecting an executable to search for shared libraries in other folders can be done using the environment variable LD_LIBRARY_PATH

To find out which requirements a library has you can run objdump -p <program>

jmcfarlan, 2016-07-11:
I downloaded the 4.4.4 version and it is closer to working.  I can run with the following to options:

  - SVDConv myDevice.svd
  - SVDConv myDevice.svd --generate=partition

However, when I run the following I get "Segmentation fault (core dumped)"

  - SVDConv myDevice.svd --generate=header

thorstendb-ARM, 2016-07-11:
Hi,
can you please attach a (small) reproducer SVD file?
Thanks!

jmcfarlan, 2016-07-12:
I am having trouble getting a smaller testcase to show the same issue.  As another data point, the svd file converts fine on the PC side using V2.86g.

Our IT folks also asked about the chances of obtaining the source code so we could compile locally on our platform.

JonatanAntoni, 2017-03-03:
@jmcfarlan Does this issue still exists with current SVDConv 3.3.1? Please review, report back or close this issue if solved, please.

jmcfarlan, 2017-03-08:
The new version fixes my issues.
Thanks.

closed 2017-03-08 by jmcfarlan