ARM-software/CMSIS_5 #64: cmsis_os2.h: non portable object IDs

ilg-ul, 2016-07-08:
All your object IDs (like `osThreadId_t`, `osTimerId_t`, etc) are defined as `uint32_t`, although all actual implementations of the `osXxxNew()` functions will return pointers, not integers.

This makes porting on 64-bits platforms impossible (or complicates the ports uselessly).

FYI, although [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) was mainly targeted to Cortex-M devices, most of the development was done on macOS, as a native POSIX process, built with the native clang compiler (64-bits), including your CMSIS RTOS validator. 

Your latest version would prevent this.

Suggestion:

- update the typedef for all osXxxId_t to `void*`
- generally do not assume the build will be done on a 32-bits platform

ilg-ul, 2016-09-14:
at my first glance at your updated documentation site I noticed that the object IDs are still `uint32_t`, which makes porting systems that use pointers to identify objects very questionable, or at least unjustifiable complicated.

if this was a deliberate decision, please re-analyse it, since it makes running tests on development environments impossible.

RobertRostohar, 2016-10-12:
Object ID typedefs have been changed to `void *`

02e45bf

ilg-ul, 2016-10-12:
super!

thank you,

Liviu

closed 2016-10-12 by ilg-ul