ARM-software/CMSIS_5 #65: CMSIS RTOSv2 API: Avoid mixing time durations (in milliseconds) with timer counts in ticks

ilg-ul, 2016-07-08:
Same as https://github.com/ARM-software/CMSIS/issues/39

ilg-ul, 2016-10-12:
@RobertRostohar, all current timeouts variables are still named `millisec`, while the manual correctly states 

> The timeout value specifies the number of timer ticks until a timeout or time delay elapses.

Suggestion:

- rename `millisec` -> `duration`

Please note that the name `ticks`, although apparently appropriate, is not generic enough, in case you'll accept multiple clocks, as used by CMSIS++, where clock can count ticks, seconds, or anything else.

RobertRostohar, 2016-10-19:
Timeouts are not tied to millisec anymore (9df0df8).

ilg-ul, 2016-10-19:
ok.

if you would define all these parameters to be of type `duration_t`, it would be even better (see https://github.com/ARM-software/CMSIS/issues/66).

ReinhardKeil, 2016-11-15:
This is done, however I understand that we need to improve the documentation

ilg-ul, 2016-11-15:
this is partly done, the parameter name was changed to `timeout`, but the types are still `uint32_t` instead of something like `osDuration_t`.

JonatanAntoni, 2017-09-01:
Just to update this discussion after a long time of inactivity:
The original issue with timeouts in ticks has been solved long time ago.

We still discuss about the benefits created by switching to strong typing (i.e. osDuration_t instead of uint32_t). Technically both results to the same behavior thus we do not have priority on this change, actually. 

On the long run I admit that using strong types instead of common ones can enhance code readability. I am going to keep this issue open until we finally decide whether to change types.

Best,
Jonatan