ARM-software/CMSIS_5 #66: CMSIS RTOSv2 API: Promote object names as first citizens

ilg-ul, 2016-07-08:
I was pleased to notice that you adopted many of the CMSIS++ suggestions, including defining object names via attributes.

It is true that the original CMSIS++ RTOS specs classified the object names as optional, so they were not defined in the function prototype, where only mandatory parameters were, but in the object attributes.

However, after using CMSIS++ for a while, I noticed that all my objects were created with the name attribute, for the simple reason that I needed to identify the object during debug sessions.

With the current trend of new debuggers, profilers, instrumentation tools, etc to provide full details about the RTOS internal objects, the object names become a real necessity.

Given this, in the current CMSIS++ RTOS specs, the object names become mandatory, and were moved in the function prototype, in the first position.

Suggestion:

- update the `osXxxNew()` functions and add a `const char* name` as first parameter

ReinhardKeil, 2016-07-12:
Names are not required for debugging and optional. Debuggers may use symbolic names in the ELF to show context.

ilg-ul, 2016-07-12:
It is correct that from a functional point of view names are not absolutely required, but the practice showed that, when available, are very useful.

CMSIS RTOS v1, via the famous macros, was able to provide names for all objects.

Symbolic names in the ELF are generally available for global objects; for inner objects, allocated on the stack, these names might not be so easily available.

Not to mention that names for C++ objects, especially those generated as a result of template expansion, may be tens of characters long, totally inappropriate for a small debugger window.

ilg-ul, 2016-10-28:
@RobertRostohar, I appreciate you took the first step in acknowledging that object names are important, by adding functions to return the thread name, timer name, etc.

however, according to the current API, names can be set only via attributes, which is very tedious.

please consider adding the name as the first parameter of all functions that create objects, as in [CMSIS++](http://micro-os-plus.github.io), this greatly simplifies usage and improves program readability.

ReinhardKeil, 2016-11-15:
We have decided not to address this as names are optional and not essential for the functionality

ilg-ul, 2016-11-15:
Names are not essential for the functionality, but they are highly recommended for debugging. 

The indirect consequence of your decision is that attributes become mostly mandatory at creation time for all objects.

ReinhardKeil, 2017-02-13:
closing as there is no further question on that

closed 2017-02-13 by ReinhardKeil