ARM-software/CMSIS_5 #70: CMSIS RTOSv2 API: Do not expect the dynamic allocator to be preferred

ilg-ul, 2016-07-08:
This is a tough one.

You made a step forward and changed the object creation method from the dreaded macros to the `osXxxNew()` method.

Since your documentation pages are not yet ready, I can only imagine how you expect them to be used.

Here is one such sample:

``` c
// This structure is port dependent, CMSIS RTOSv2 currently does not define it.
port_event_flags_t evf_cb;

osEventFlagsAttr_t evf_attr;
memset(evf_attr, 0, sizeof(osEventFlagsAttr_t)):
evf_attr.name = "evf";
evf_attr.cb_mem = &evf_cb;
evf_attr.cb_size = sizeof(evf_cb);

osEventFlagsId_t evf;
evf = osEventFlagsNew(evf_attr);
```

Please compare the equivalent CMSIS++ solution (the C API):

``` c
os_evflags_t evf;
os_evflags_create(&evf, "evf", NULL);
```

Don't you think your solution is a bit excessive?

Please correct me if I'm wrong, but I see no other solution for creating objects that are not dynamically allocated, and this one is not ok.

If you assumed that most of the objects will be dynamically allocated, this is not ok either.

ReinhardKeil, 2016-11-15:
I think the documentation makes this clear now
http://arm-software.github.io/CMSIS_5/RTOS2/html/MemoryAllocation.html

ilg-ul, 2016-11-15:
1. your link refers to RTX, not to CMSIS RTOSv2. are these memory management features mandatory for all implementations?

2. the example in that page refers to `os_thread_t`, which is not defined in `cmsis_os2.h`, thus your example is not portable.

3. the naming convention of `os_thread_t` (although identical to the one used in CMSIS++, which use lower case names), does not comply to your CamelCase naming convention, used in CMSIS RTOSv2.

ilg-ul, 2016-11-15:
although not directly related to this topic, another non conform type name is

- `os_thread_func_t`.

could you review the naming convention in `cmsis_os2.h` and fix all inconsistencies?

ReinhardKeil, 2016-12-07:
internal names are now changed to osRtxXxxx

closed 2016-12-07 by ReinhardKeil