ARM-software/CMSIS_5 #71: CMSIS RTOSv2 API: Should one thread be allowed to osThreadFlagsClear () of another thread?

ilg-ul, 2016-07-09:
The current prototype of `osThreadFlagsClear()`

``` c
int32_t osThreadFlagsClear (osThreadId_t thread_id, int32_t flags);
```

This means any thread can clear the flags of any other thread. 

While working on the CMSIS++ RTOS API I also had to make such decisions. In this case my resolution was that threads should be allowed only to clear their own flags.

So in [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) the `flags_clear()` function was moved from the `thread` class to the `this_thread` namespace.

The current C++ definition is:

``` c++
    namespace this_thread
    {
      result_t
      flags_clear (flags::mask_t mask, flags::mask_t* oflags);
    }
```

and the C version is:

``` c
os_result_t
os_this_thread_flags_clear (os_flags_mask_t mask, os_flags_mask_t* oflags);
```

Suggestion:

- change the prototype to something like 

``` c
int32_t osThisThreadFlagsClear (int32_t flags);
```

Similarly for osThreadFlagsGet():

``` c
int32_t osThisThreadFlagsGet (void);
```

ReinhardKeil, 2016-09-29:
I would argue that our implementation is more orthogonal and provides additional features.

ilg-ul, 2016-09-29:
the basic principle behind designing APIs is "make interfaces easy to use correctly and hard to use incorrectly"

thread flags are some kind of binary semaphores owned by the thread, raised by other threads/irqs and, being owned by the thread, must be consumed only by the thread. 

when you enable other threads to clear flags, you simply interfere with this mechanism and artificially consume events.

this does not make your design easy to use correctly and hard to use incorrectly.

rizickus, 2016-09-30:
Suggestion: make two different API calls. Add `osThisThreadFlagsClear` and leave `osThreadFlagsClear`.

I agree with ilg-ul that `osThreadFlagsClear` can lead to really bad system designs but having ability to use generic function is preferable. There are cases where you should avoid clearing flags of other threads but also there should be some cases where it fits. OS level API might be backbone for your application, it should not limit you.

ilg-ul, 2016-09-30:
> but also there should be some cases where it fits

yes, everything is possible, but I would say that if this happens, it is a good sign of a poor design, and the API should help prevent poor designs.

thread flags should be used for synchronisation only, i.e. to wakeup a thread waiting for a condition. in a safe design, the notified thread should not rely on the flags alone, but should check the condition.

this is actually true for any synchronisation primitive, after it returns, the application should check if the condition is true, since it might be very well a timeout, a cancel issued by another thread, or even an erroneous thread.resume() coming from the dark.

given the above, the originator of the event can only notify the thread to make it check for the condition. if the originator detects that the condition is no longer true it should not bother to notify the thread that it changed its mind, because the thread will wakeup, check the condition, find it false, and loop.

the manual should clearly explain the above. in a normal use case, with auto-clear, none of these two functions is needed. without auto-clear, the waiting thread should clear the flags, and this is the only reason for `osThisThreadFlagsClear()` to exist.

if you choose to keep `osThreadFlagsClear()`, you should clearly explain why it should not be used by other threads/irqs (which will probably raise the question of why is the function present).

ilg-ul, 2016-09-30:
FYI, there are several other calls that should be tied to the current thread only.

in [CMSIS++](http://micro-os-plus.github.io), these calls are grouped in the [`os::rtos::this_thread`](http://micro-os-plus.github.io/reference/cmsis-plus/namespaceos_1_1rtos_1_1this__thread.html) namespace and currently include:

- suspending a thread (see #69)
- exiting a thread
- yielding a thread
- waiting for thread event flags
- getting thread flags
- clearing thread flags

RobertRostohar, 2016-10-12:
`osThreadFlagsClear` and `osThreadFlagsGet` functions have been limited to current running thread.

461d000

ilg-ul, 2016-10-12:
>  limited to current running thread

that's good. 

it would be even better if you would rename all thread related functions that do not have a thread_id, i.e. apply to the current thread, as `osThisThreadXyz()`; this explicit naming would make the use unambiguous.

the ISO C++ standard uses a separate namespace for this purpose (`this_thread::`), and [CMSIS++](http://micro-os-plus.github.io/) takes the same approach for the RTOS C++ API. the CMSIS++ RTOS C API uses a similar naming convention, prefixing the functions with [`os_this_thread_`](http://micro-os-plus.github.io/reference/cmsis-plus/group__cmsis-plus-rtos-c-thread.html).

ilg-ul, 2016-10-12:
@RobertRostohar, please see #69 too, it is somehow related.

ReinhardKeil, 2016-11-15:
We have discussed that, but believe that our naming conventions are easier to navigate.  All thread related functions start with osThread.

ReinhardKeil, 2016-12-07:
closing as there is no further comments

closed 2016-12-07 by ReinhardKeil