ARM-software/CMSIS_5 #73: CMSIS RTOSv2 API: Separate blocking from non-blocking calls

ilg-ul, 2016-07-12:
In ARM CMSIS RTOS v1, most blocking functions, if called with a 0 timeout, changed behaviour to a non-blocking call, and returned immediately.

Although functional, this approach is not very fortunate, and my suggestion is to separate blocking from non-blocking calls.

From example, now you have:

``` c
void *osMemoryPoolAlloc (osMemoryPoolId_t mp_id, uint32_t millisec);
```

To this I would add:

``` c
void *osMemoryPoolTryAlloc (osMemoryPoolId_t mp_id);
```

This not only increases readability, since the `Try` word in the name is much more intuitive than the `0` timeout, but also makes the implementation cleaner, since now you can validate that the `Alloc` function is not called from a handler, but allow the `TryAlloc` function to be called from a handler.

In POSIX, most functions have 3 variants, (infinite block, timeout block, try), and [CMSIS++](http://micro-os-plus.github.io/cmsis-plus/) uses the same approach, but it is debatable if the infinite block is really needed, since it can be implemented with a FOREVER value for the timeout. However, the try version is really distinct, and in CMSIS++ it does have a distinct, fast implementation.

Suggestion:

- add separate `Try` versions for all blocking calls, and, where possible, explicitly mark them as callable from ISRs, while the blocking versions must be explicitly marked as non-callable from ISRs.

See also: https://github.com/ARM-software/CMSIS/issues/45

ilg-ul, 2016-10-28:
@RobertRostohar?

ReinhardKeil, 2016-11-15:
We have decided to solve this problem via the timeout value
http://arm-software.github.io/CMSIS_5/RTOS2/html/CMSIS_RTOS_TimeOutValue.html

ReinhardKeil, 2016-12-07:
closing now as there is no further activity

closed 2016-12-07 by ReinhardKeil