ARM-software/CMSIS_5 #75: Issues and improvment in current Beta

CanastraRF, 2016-08-14:
Issues and improvment in current Beta

Dear community
I download CMSIS 5 Beta 10 and test it against our coding standard.
It would be nice if this code complies more with the MISRA rules and other coding standards.

With the following improvement the code complies better to the MISRA rules:

1)
A "U" suffix shall be applied to all integer constants that are represent in an unsigned type. (MISRA C 2012 Rule 7.2)

Example
const unsigned int a[2U] = { 12345U, 0x1234U };

2)
Add as much as possible const to all function parameter and member function in classes.

A pointer should point to a const qualified type whenever possible (MISRA C 2012 Rule 8.13)

Example
void init(int const i);
void get(int * const pi);
void set(int const * const pi);

class C
{
void set(int * const pi) const;
};

3)
In C++ C-style casts (other than void casts) and functional notation casts (other than explicit constructor calls) shall not be used. (MISRA C++ 2008 Rule 5-2-4)

4)
In RTOS source I think it is dangeros to use new as a member of OS_TSK. In C++ new is a reserved keyword.

5)
GetEnableIRQ()
In Core/Include the new API NVIC_GetEnableIRQ() return 0 (Interrupt is not enabled) for negative IRQn. But most of the negative IRQn are non maskable IRQ and can't be disabled.
Please return 1 for all negative IRQn.

6)
RTOS2
I can't build RTOS2 with ARMCC 5.06U2. There are lot of error in core_cm.h.

I've modified lot of the above cases in Core/Include, Driver, RTOS
My version is attached to this case.

Reto Felix

[V5Beta10.zip](https://github.com/ARM-software/CMSIS_5/files/417088/V5Beta10.zip)

franzhollerer-Artesyn, 2016-08-15:
I am not working for Keil, nor am I involved in the CMSIS development. Therefore this post only reflects my personal meaning.

First off all MISRA allows some relaxation on 3rd party code as the CMSIS library. Therefore I think there is no need to worry about MISRA compliance too much.

Now to the technical aspects.

1)
> const unsigned int a[2U] = { 12345U, 0x1234U };

I agree that it is good practice to add the "U" to the initializer list.  But the risk is minimal. A modern compiler will generate a warning if there is a change of sign:

>  warning:  #68-D: integer conversion resulted in a change of sign

In contrast I lack some understanding why

> const unsigned int a[2U];

should be more safe / less error prone than:

> const unsigned int a[2];

What is the rational to apply MISRA rule 7.2 here?

To go a step further the C99 standard specifies that the expression which specifies the size of an array shall have an *integer* type. It is not necessary an *unsigned* type. Therefore, I think rule MISRA rule 7.2 does not apply here.

> A pointer should point to a const qualified type whenever possible (MISRA C 2012 Rule 8.13)
>
> Example
>  void init(int const i);
>  void get(int * const pi);
> void set(int const * const pi);

MISRA rule 8.13 requires that "A pointer should point to a const-qualified type whenever possible". It does not require the pointer itself to be const qualified. And as the pointer is passed by value, it would not make much sense anyway.

>  void init(int const i);

The variable *i* is passed by value. It is a copy. It is not possible for  init() to change the original variable *i*, even without *const* qualifier. Thus what is the benefit to add the *const* qualifier here?

>  void get(int * const pi);

The pointer *pi* is passed by value. It is a copy. The original pointer cannot be changed by the function. Thus what is the benefit of adding the const qualifier here?

> void set(int const * const pi);

Again, *pi* is passed by value.

To my understanding 

> void set(int const * pi);

is sufficient to comply to MISRA rule 8.13.

> class C
>  {
>  void set(int * const pi) const;
>  };

Same argument as above. Furthermore, I wonder how a setter member function can be const.

CanastraRF, 2016-08-16:
Thank you for your comment.

We are using a static analyzing tool with our rule set to check our project. The tool setup is much simpler when all files comply with the rules. Otherwise we must set an exception for lot of files.
I think we are not the only company that checks the code against the MISRA rules. So this can help other developer too.

1)
I know that most compiler didn't make any difference but I see, that some conditional breaks change when the literal is unsigned.

Our rule set requires a 'U' to all unsigned. The modification in the source is done once.

2)
const can help developer to reduce bugs.

Example:
```
int * p

(*p)++;
//is different from
*(p++);
```
and lot of developers didn't know what happened when the see 
```
*p++;
```
When the value (int const *) or the pointer (int * const) is const only one of the above statement are allowed. The other result in a compile error. (If both are const then both statement are not allowed).

When we transfer objects to a function by pointer it helps when the pointer is const. 

Example:
```
typedef struct 
{
	int i
} ts);
void func(ts * const p)
{
	p++;	// is not allowed.
	p->i++;	// is allowed.
}
```
inside classes const member function can be used to const objects and const object can placed in flash.

The resulting code with or without const is mostly the same.
So const help only the developer to reduce bugs.

Reto Felix

franzhollerer-Artesyn, 2016-08-17:
@CanastraRF 
I see your arguments, even I don't fully agree with the conclusion. I prefer to address this with education, rather than enforcing rules. It's counterproductive if engineers follow rules they don't grasp, e.g. because they lack an understanding of the C language. I have seen very strange outcome from rule driven engineering...

I am not against MISRA. I think every firmware engineer should be familiar with the MISRA standard and understand the rational behind the rules. But I don't like how MISRA is applied in practice.

Anyway, this is my personal opinion. I want to come back to your examples in point 2) 

> A pointer should point to a const qualified type whenever possible (MISRA C 2012 Rule 8.13)
>
> Example
> void init(int const i);
> void get(int * const pi);
> void set(int const * const pi);

I absolutely agree with MISRA Rule 8.13 and also agree with you that every developer should care about const correctness. And I also agree that the correct use of the const qualifier can help to avoid bugs and make an API more precise.

But I think your examples are inapplicable to argument for MISRA Rule 8.13.

Let us consider strcpy():

```c
char *strcpy(char *dst, char *src);
```

To comply with MISRA Rule 8.13: "A pointer should point to a const-qualified type whenever possible".

the function signature should be:

```c
char *strcpy(char *dst, const char *src);
```

Or to stick with your syntax:
```c
char *strcpy(char *dst, char const *src); 
```

I prefer the first variant. I find it more readable and it is more widespread than the second variant. You find it in POSIX (http://pubs.opengroup.org/onlinepubs/9699919799/functions/strcpy.html) and in the MISRA examples. But this is a personal preference.

You can claim violation of MISRA Rule 8.13 if the above const qualifier is missing, i.e. the dereferenced src is not a const char.

But the pointer *src* itself to be const qualified is not covered by MISRA Rule 8.13. Your own coding standard may encourage:

```c
char *strcpy(char *dst, char const * const src); 
```
But I think you should not argument with compliance to MISRA Rule 8.13 in this case. You can argument with your own best practice experience, but not to align the code better with MISRA.

Right? Have I overseen something?

Beside that, I agree that CMSIS should care about const correctness, especially in the spirit of MISRA Rule 8.13. 

I am not convinced of the pointer *src* to be const. It is passed by value and therefore a copy. Why shouldn't it be used within the implementation to iterate through the string? It is a case by case decision. Nothing to enforce by a rule checker. And nothing to argument with MISRA.

ilg-ul, 2016-08-17:
> Or to stick with your syntax:
> char *strcpy(char *dst, char const *src); 

although this is not covered by the ISO/IEC standard or by MISRA, the easiest way to remember the const/volatile semantic is the rule of thumb **const makes a constant whatever is on its left**. 

another such rule of thumb is that these types should be read in reverse order, from right to left.

as such, `char const *` is read as _pointer to constant character_, which is quite natural.

this is obviously a matter of personal preferences, but, given these rules, this syntax seems more consistent.

ReinhardKeil, 2016-09-05:
@CasandarCF 
we are using PC-LINT for Misra checking.  We are aware that this tool is not perfect.  I would be interested to know which tool you are using.  I would like to understand where we are violating the MISRA C 2012 Rule 7.2 or MISRA C++ 2008 Rule 5-2-4. Can you give some examples?

As CMSIS is an API definition we would restrict implementations if we enforce this rule:
"A pointer should point to a const qualified type whenever possible (MISRA C 2012 Rule 8.13)"
See also the discussion with franzhollerer-Artesyn.

Item 4:
I agree with you on item 4, however RTX v4 is a code base that is stable and well tested and the source code that is using 'new' is in C source.  In RTX v5 we are not using 'new' as member variable. 

Item 5:
GetEnableIRQ() will be reviewed based on your comment.

Item 6:
RTX5 is developed currently with ARM Compiler 6 and GCC.  Once development is complete, we may consider porting to ARM Compiler 5, but a pre-compiled library would work with both ARM Compiler 5 and ARM Compiler 6 (so the priority for porting to ARM Compiler 5 is not high).

CanastraRF, 2016-09-06:
Dear ReinhardKeil

We are using Pc-Lint too. At the moment we are testing pclint plus beta 8.
PcLint support much more rules the only MISRA checking. 
Purhaps our settings reports more warnings the only MISRA C / C++.

I add a zip with modified files to this issue. So you can diff the current version with my modification.

I've added some const and 'U' to:
- cmsis\core\include\core_*.h
- cmsis\driver\include\driver_*.h
- cmsis\driver\drivertemplates\driver_*.h
- cmsis\rtos\rtx\inc\*.h
- cmsis\rtos\rtx\src\*
- cmsis\rtos\rtx\usercodetemplates\*
- device\arm\armcm?\include\*.h
- device\arm\armcm?\source\*.c

I've replaced some c-cast with *_cast<>() in:
- cmsis\rtos\template\cpp\

some messages from PcLint 9.0L

cmsis_armcc.h(52): Note 953: Variable '__regControl' (line 50) could be declared as const

cmsis_armcc.h(64): Note 952: Parameter 'control' (line 60) could be declared const [MISRA C++ Rule 7-1-1]

core_cm4.h(1883): Note 9027: Unpermitted operand to operator '<<' [MISRA 2012 Rule 10.1, required]
core_cm4.h(1883): Info 701: Shift left of signed quantity (int)
core_cm4.h(1883): Note 9027: Unpermitted operand to operator '&' [MISRA 2012 Rule 10.1, required]
core_cm4.h(1883): Note 9029: Mismatched essential type categories for binary operator [MISRA 2012 Rule 10.4, required]

core_cm4.h(2004): Info 717: do ... while(0);
core_cm4.h(2004): Note 9036: Conditional expression should have essentially Boolean type [MISRA 2012 Rule 14.4, required]

core_cm4.h(2144): Note 9029: Mismatched essential type categories for binary operator [MISRA 2012 Rule 10.4, required]
core_cm4.h(2144): Info 737: Loss of sign in promotion from int to unsigned int

core_cm4.h(1980): Note 954: Pointer variable 'vectors' (line 1978) could be declared as pointing to const [MISRA 2012 Rule 8.13, advisory]

Puthaps You can call me by phone. For me it's much easiers to discuss in German then English. The keil support has got my identity.

Reto Felix

ReinhardKeil, 2016-09-12:
MISRA 2012, Rule 8.13 applies to pointers (and arrays which are another form of pointers), not to scalar type parameters.

This is therefore not a MISRA requirement:
void init(int const i);

CanastraRF, 2016-09-13:
@ReinhardKeil 
MISRA C++ 2008 Rule 7.1.1 require:
A variable which is not modified shall be const qualified.

So PC-Lint reports 

Note 952: Parameter could be declared const [MISRA C++ Rule 7-1-1]

if a variable is not modified inside a function.

ReinhardKeil, 2016-09-13:
@CanastraRF thanks for that feedback. After discussions here, we have concerns that implementing these C++ rules will create warnings at some C compilers. As you know, there are differences in the behaviour of 'const' between C and C++ and CMSIS supports a diverse range of C and C++ compilers.

Instead we will document which standards and rules we follow with CMSIS.

CanastraRF, 2016-09-13:
@ReinhardKeil 
Thank You for your explanation.
Can you list which compiler will create warnings?

Otherwise you define a symbol like _CONST and set it to const or empty depending of the compiler.

ReinhardKeil, 2016-11-15:
We will investigate further. However as it is C code, we will apply MISRA-C 2012 rules.

JonatanAntoni, 2017-02-27:
Regarding NVIC_GetEnableIRQ, please note that these function is not meant to be called for system interrupts, see [1] states "IRQn must not be negative".

[1] https://www.keil.com/pack/doc/cmsis/Core/html/group__NVIC__gr.html#ga72f102d31af0ee4aa7a6fb7a180840f3

JonatanAntoni, 2018-01-09:
In the meanwhile we made some progress regarding MISRA compliance, mainly for RTX5. Please find the decisions made in the [CMSIS documentation](file:///C:/Users/jonant01/git/CMSIS_5/CMSIS/Documentation/RTOS2/html/misraCompliance5.html).

JonatanAntoni, 2018-03-20:
We have updated many places all over CMSIS to be better MISRA compliant in the meanwhile. May I ask you all to provide some up-to-date feedback, please? I think we should be able to close this issue.

ReinhardKeil, 2018-05-23:
RTX is now MISRA-2012 complaint - deviations are documented.

JonatanAntoni, 2018-06-22:
Closing this today as we made major progress since this issue was raised. Please feel free to raise additional issues (or preferably pull requests) for further enhancements. Thanks to all for your contribution.

closed 2018-06-22 by JonatanAntoni