ARM-software/CMSIS_5 #76: CMSIS RTOSv2-API: Testing the example

FerdinandEn, 2016-08-24:
Dear Community,

I did some experiments using RTOS v2 API beta 11.

Looking at the example RTX5_Blinky I have the following problems:

To show RTOS information you are not longer using "System and Thread Viewer" but Component Viewer.

In Component Viewer "Stack Overrun check" is displayed as Disabled. In RTX_Config.c Stack overrun checking is selected. De-selecting check overrun in RTX_Config.c leads to a link error

.\Flash\Blinky.axf: Error: L6200E: Symbol os_ThreadStackCheck multiply defined (by rtx_thread.o and rtx_config.o).

Is it possible to display a Thread name in component viewer (like Idle thread)?

I have tried to define osThreadAttr_t. Prioity is changed, but the name is not displayed. 

In which way I have to handle osThreadAttr_t-variable. Has this variable to be global or static?
Is it sufficient that the elements  cb_mem and stack_mem are global or static?

Event Viewer does not display RTOS state. I did not see ITM-msgs in RTOS code. Is Event viewer usable in future releases?

In you example you are using thread flags, e. g. in thread clock

osThreadFlagsWait(0x0100, osFlagsWaitAny, osWaitForever);  

Looking in the manual osThreadFlagsWait() does not consume the flag. So using osThreadFlagsWait(0x0100, osFlagsWaitAny, ..) the flag 0x0100 should be still set. To consume the flag I would expect a code like

osThreadFlagsWait(0x0100, osFlagsWaitAny+osFlagsAutoClear , osWaitForever); 
or
osThreadFlagsWait(0x0100, osFlagsWaitAny|osFlagsAutoClear , osWaitForever); 

I would be very pleased, if you could give me the information mentioned above.

Some remarks to the pack creation. The current files does not work without modifications (doxgen 1.8.11)

gen_pack.bat line 102 - 104

Doxygen documentation is produced and deleted before Pack is build

I have commented the lines

::PUSHD ..\Documentation
::FOR %%A IN (Core, DAP, Driver, DSP, General, Pack, RTOS, RTOS2, SVD) DO IF EXIST %%A (RMDIR /S /Q %%A)
::POPD

ARM.CMSIS.pdsc

I have to rename line 1364 rtx_implementation.html to rtxImplementation.html

Doing this modifications I can produce the pack

I have an additional question. I am starting a development using STM32F746.

Using RTOS V1 I have to include

#ifdef RTE_CMSIS_RTOS_RTX
extern uint32_t os_time;

/**
  * @brief Function to implement HAL-Wait function
  */
uint32_t HAL_GetTick(void) { 
  return os_time; 
}
#endif

This is possible, because RTOS is started by __main().

Using RTOS V2 a possible solution is to use

uint32_t HAL_GetTick(void) { 
  return osKernelGetTime(); 
}

Doing this, will lead to Hardfault calling SystemClock_Config()

int32_t main(void)
{ 
  // ------- Start of Standard initialization ------------------
  MPU_Config();          /* Configure the MPU           */
  CPU_CACHE_Enable();    /* Enable the CPU Cache        */
  HAL_Init();            /* Initialize the HAL Library  */
  SystemClock_Config();  /* Configure the System Clock  */
  
  SystemCoreClockUpdate();
  osKernelInitialize();  
  osThreadNew(mainThread, NULL, NULL);
  osKernelStart();
  
  for(;;)
  {
  }
}  
  
A solution could be a code like this

uint32_t HAL_GetTick(void) { 
  if(osKernelGetState() == osKernelRunning)
  {
     return osKernelGetTime(); 
  }
  else
  {
    return(0);
  }
}

The code requires  a call of osKernelGetState() for each HAL_GetTick() call.

Do you have another solution?

At the moment I am using RTOS V1 and CMSIS DSP in my lessons. For the exam I am delivering a pdf manual for these parts. I am using Latex to write the documents. 

Writing this manual would be simplified, if you could include an additional doxygen line in the code, if you include images, e.g.

  * @image html speedy.png
  * @image latex speedy.png "Roboter Speedy" width=7cm

FerdinandEn, 2016-09-28:
Using the latest Version of CMSIS_5 Component Viewer RTX5.scvd does not longer work.

`Expression: 'TCB[i].stack_val  = __CalcMemUsed (TCB[i].stack_mem, TCB[i].stack_size, 0xCCCCCCCC, 0xE25A2EA5);
          TCB[i].stack_over = TCB[i].stack_val >> 31;
          TCB[i].stack_maxb = TCB[i].stack_over ? TCB[i].stack_size : TCB[i].stack_val & 0xFFFFF;
          TCB[i].stack_maxp = TCB[i].stack_over ? 100 : (TCB[i].stack_val >> 20) & 0x1FF;'
E306: Execution problem - __CalcMemUsed(): max. size exceeded (max.:=65536, cur:=536877064)
Expression: 'TCB[i].stack_cur  = TCB[i].stack_mem + TCB[i].stack_size;
          TCB[i].stack_cur -= sp;
          TCB[i].stack_cur *= 100;
          TCB[i].stack_cur /= TCB[i].stack_size;'
E201: Division by zero - operator '/'`

RobertRostohar, 2016-09-30:
RTX5.scvd has been updated and should resolve the mentioned problem.

eugene345, 2016-10-12:
I'm playing with RTX5_Blinky example and see similar problem
This is before stepping over `tid_phaseA = osThreadNew(phaseA, NULL, NULL);` :

[image]

and this is after:

[image]

eugene345, 2016-10-12:
Sorry, seems I used old RTX5.scvd file
There is no such problem with the newest version.

ReinhardKeil, 2016-11-15:
This seems to be fixed now. Can you confirm and close the issue.

closed 2016-11-30 by FerdinandEn