ARM-software/CMSIS_5 #78: CMSIS DSP arm_var_f32.c

dcurrie, 2016-08-30:
The algorithm used for arm_var_f32 has poor numerical stability. See this blog page for an explanation: http://www.johndcook.com/blog/standard_deviation/ and this page for comparisons of the alternatives: http://www.johndcook.com/blog/2008/09/26/comparing-three-methods-of-computing-standard-deviation/

Implementing the recommended Welford method (without regard for performance tuning) gave me an additional decimal digit of accuracy in my application relative to the CMSIS version. See attached.
[arm_var_f32_e.c.txt](https://github.com/ARM-software/CMSIS_5/files/445806/arm_var_f32_e.c.txt)

It may be that the third approach, the "direct method" using two passes over the data, also has good accuracy, but performance will likely be dominated by cache effects. In either case, Welford or direct method, performance testing and tuning are warranted.

GuentherMartin, 2016-12-22:
Hello dcurrie.

file arm_var_f32.f is updated.
Please see https://github.com/ARM-software/CMSIS_5/commit/61d84104b3bece5c6bbb8b934c1a8b2851e4c314

dcurrie, 2016-12-22:
The new version passes the unit tests in my application, whereas the previous version did not have sufficient accuracy. Thank-you for the fix, I am closing this issue.

closed 2016-12-22 by dcurrie