ARM-software/CMSIS_5 #79: Bug in core_cm*.h

CanastraRF, 2016-09-01:
Dear Community
When I create a simple C++ project with the new CMSIS 5 Beta I get error 
Main.cpp(4): error:  #369-D: variable "ArmCpuDef" has an uninitialized const or reference member

Attached is my demo project.

When I modify core_cm*.h I can fix this.

[UV.zip](https://github.com/ARM-software/CMSIS_5/files/449018/UV.zip)
[UV.zip](https://github.com/ARM-software/CMSIS_5/files/449037/UV.zip)
Replace 

#define     __IM     volatile const      /*! Defines 'read only' structure member permissions */

with

#ifdef __cplusplus
  #define   __IM     volatile            /*! Defines 'read only' structure member permissions */
#else
  #define   __IM     volatile const      /*! Defines 'read only' structure member permissions */
#endif

Please fix this
Reto Felix

ReinhardKeil, 2016-09-05:
There is no need to define
<ARM_CPU_SYS_TypeDef	sArmCpuDef;>

The usage of the peripheral definitionss is:
<
// NOTE: these defines are already part of ARMCM4.h, just repeated to be complete
#define ARM_CPU_CFG_BASE          (0xDFFF0000UL)
#define ARM_CPU_SYS_BASE          (ARM_CPU_CFG_BASE  + 0x00000UL)
#define ARM_CPU_SYS        ((ARM_CPU_SYS_TypeDef *) ARM_CPU_SYS_BASE)

int main(void)
{
	ARM_CPU_SYS->CLKCFG = 0;
}
>

CanastraRF, 2016-09-05:
But when I Need a backup of a current struct I've got the same problem.

``#include <string.h>
#include "ARMCM4.h"

ARM_CPU_SYS_TypeDef	sArmCpuDef;

int main(void)
{
	ARM_CPU_SYS->CLKCFG = 0;
	::memcpy(&sArmCpuDef, ARM_CPU_SYS, sizeof(sArmCpuDef));

	return 0;
}
``

closed 2016-09-13 by ReinhardKeil