ARM-software/CMSIS_5 #83: CMSIS RTOSv2 API: osThreadAbortWait() vs osThreadResume() and WAITING vs SUSPENDED

ilg-ul, 2016-09-15:
The documentation explains that `osThreadAbortWait()` resumes threads in WAITING mode and `osThreadResume()` resumes threads in SUSPENDED.

This makes resuming a thread slightly complicated, since a call to `osThreadGetState()` is needed to know the current state.

Personally I think that the distinction between `SUSPENDED` and `WAITING` is not relevant, a live thread is either _READY_ or _not READY_. In [CMSIS++](http://micro-os-plus.github.io) I call the _not READY_ state `SUSPENDED`. For the user it is not relevant why the thread is not ready, if the user want the thread to run, issuing `osThreadResume()` is a natural choice.

Even more, the portable API used internally between the CMSIS++ portable synchronisation and communication objects and the scheduler uses exactly these two functions, `thread.resume()` and `thread.suspend()`, because the scheduler itself does not need to know anything more, the thread is either in the READY list, or not, and in this case it is the responsibility of the synchronisation object(mutex, semaphore, etc) to keep track of the thread in various lists, totally independent from the scheduler (with the µOS++/CMSIS++ implementation, it is currently possible to use the mutex, semaphore, etc from the portable CMSIS++ implementation and the FreeRTOS scheduler).

Suggestion:

- simplify things, and remove the `WAITING` state; the `SUSPENDED` state is enough
- remove the `osThreadAbortWait()` function; `osThreadResume()` is enough

RobertRostohar, 2016-10-12:
`osThreadAbortWait()` function has been removed and states `WAITING` and `SUSPENDED` have been joined.

3ade623

ilg-ul, 2016-10-12:
super!

closed 2016-10-12 by ilg-ul