ARM-software/CMSIS_5 #85: CMSIS-Driver: clang build fails due to bad macro name

ilg-ul, 2016-10-01:
My attempt to compile files which include CMSIS-Driver fail, for example:

```
/Users/ilg/My Files/MacBookPro Projects/uOS/micro-os-plus-iii-tree.git/ilg/arm/arm-cmsis-xpack/CMSIS/Driver/Include/Driver_USART.h:57:9: error: macro name is a reserved identifier [-Werror,-Wreserved-id-macro]
#define __DRIVER_USART_H
```

To reproduce the problem, use `-Wall -Werror -Weverything`.

Suggestion:

- rename the guard macros used in all header file; avoid initial underscores; instead, add one final underscore, in this case `DRIVER_USART_H_`

ilg-ul, 2016-10-02:
please see the patched version used in [CMSIS++](http://micro-os-plus.github.io/):

- https://github.com/xpacks/arm-cmsis/blob/xpack/CMSIS/Driver/Include/Driver_USART.h

ilg-ul, 2016-10-12:
@RobertRostohar, the `cmsis_os2.h` header has the same problem, please update the name of the guard macro, for example to `CMSIS_OS2_H_`.

RobertRostohar, 2016-10-13:
Guard macros have been updated in Drivers and RTOS2 as suggested.

4611e7f, 785c2aa

ilg-ul, 2016-10-13:
great, thank you.

closed 2016-10-13 by ilg-ul