ARM-software/CMSIS_5 #87: RTOSv2 os_Info not initialized with correct values

FerdinandEn, 2016-10-04:
I have tried to write a program for STM32F746 using System Workbench for STM32 (gcc)

A call of
`osKernelInitialize();`
fails (-1)

The reason is, that os_Info is not initialized with valid content.

I attach the project

[RTOSv2test.zip](https://github.com/ARM-software/CMSIS_5/files/507779/RTOSv2test.zip)

RobertRostohar, 2016-10-04:
os_Info is put into a special section ".os.data".

When using gcc the linker script needs to be updated. Add the following into ".data:" after `*(.data*)`:
`*(.data*)`
`*(.os.data*)`
`*(.os.object*)`

RobertRostohar, 2016-10-10:
Sections have now been renamed to start with ".bss" /  ".data" which should work with default gcc linker scripts.

ReinhardKeil, 2016-10-11:
closing issue as it appears to be resolved.

closed 2016-10-11 by ReinhardKeil