ARM-software/CMSIS_5 #92: No thread delete function in RTOS2

joabeck, 2016-10-18:
This is possibly only a question and not an issue.

For all other RTOS resources there exists a New-function and a Delete-function, but not for threads. This seems a little bit inconsequent.
When will the memory for a thread be free again, after threadExit or threadTerminate?

Thanks
Joachim

ilg-ul, 2016-10-18:
for threads we should distinguish the memory required for the thread control block and the stack.

normally threadTerminate() should be used for a different thread (this restriction should be mentioned in the manual), and in this case the thread stack can be fully freed when terminate() returns.

for threadExit() things are a bit more complicated, since this function is called on the context of the current thread, which means the current stack cannot be freed while executing thread code, so the thread should be added to a list of threads to be deleted, and the list should be iterated by another thread and terminated threads will be finally deleted (usually this happens on the idle thread, but any other strategy is ok).

the conclusion is that threads exiting themselves may take a while to be deleted.

regarding the delete function, for consistency reasons, the [CMSIS++ RTOS C API](http://micro-os-plus.github.io/reference/cmsis-plus/group__cmsis-plus-rtos-c-thread.html) defines a `os_thread_delete()`, which deallocates the thread control block allocated by `os_thread_new()`.

actually CMSIS++ defines two pairs of functions for each object, for example:

```
void os_thread_construct (os_thread_t* thread, const char* name, os_thread_func_t func, const os_thread_func_args_t args, const os_thread_attr_t* attr);
void os_thread_destruct (os_thread_t *thread);
```
and
```
os_thread_t* os_thread_new (const char* name, os_thread_func_t func, const os_thread_func_args_t args, const os_thread_attr_t* attr);
void os_thread_delete (os_thread_t *thread);
```

the construct/destruct are exactly the C++ constructor/destructor, while new/delete are exactly the C++ new/delete, which, in addition to construct/destruct, allocate/deallocate memory for the thread control block.

this highly consistent mechanism is used for all CMSIS++ system objects.

joabeck, 2016-10-22:
Ok now I know how this is solved in CMSIS++. But how is it handled in CMSIS 5. This is still unclear to me.
Thanks
Joachim

ReinhardKeil, 2016-11-15:
CMSIS-RTOS2 has a thread attribute *osThreadJoinable* (see http://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html). When *osThreadJoinable* is not set *osThreadTerminate* instantly deletes the thread. When *osThreadJoinable* is set, you must use *osThreadJoin* to release the resources of the thread.

ReinhardKeil, 2017-02-13:
closing as there is no further question on that

closed 2017-02-13 by ReinhardKeil