ARM-software/CMSIS_5 #94: RTOS2 uVision crash using osThreadAttr_t

FerdinandEn, 2016-10-21:
[03-ThreadDefine.zip](https://github.com/ARM-software/CMSIS_5/files/544423/03-ThreadDefine.zip)
Using osThreadNew() with osThreadAttr_t let uVision crash.

I have attached the example design.
[03-ThreadDefine.zip](https://github.com/ARM-software/CMSIS_5/files/544427/03-ThreadDefine.zip)

RobertRostohar, 2016-10-24:
Could you provide a bit more information how to provoke the crash and which uVision version you have ?

I was able to duplicate one crash by using the provided example following these steps:
- build target
- start debug session
- run target and stop
- open View - Watch Windows - RTX RTOS

The crash occurred with uVision 5.21.1.0. It does not occur with newer uVision version so that particular problem has already been resolved.

FerdinandEn, 2016-10-24:
I am using uVision V5.21.1.0 (MDK 5.21a)

uVision crash in Debugger after starting the MC-program, if RTX Component Viewer is open.

RobertRostohar, 2016-10-24:
Thanks for the additional info.

The problem should be resolved in the latest uVision (will be part of the next MDK release).

ReinhardKeil, 2016-11-15:
MDK 5.22 is fixing this problem.

ReinhardKeil, 2016-11-17:
closing this issue now as there is no further activity

closed 2016-11-17 by ReinhardKeil