ARM-software/CMSIS_5 #100: RTOS2 documentation - thread and message queue

FerdinandEn, 2016-10-27:
Due to the bug in MDK 5.21a it is difficult for me to test thread settings using osThreadAttr. In the documentation some examples are given. I would be very pleased, if you could give some additional examples to cover some cases.

**RTOS2 documentation - thread**

Case Thread which does not use default stack size (dynamic memory allocation)
```
const osThreadAttr_t thread1_attr = {
  .stack_mem  = NULL
  .stack_size = 800
}
```
With the definition above, it should be possible to define a stack in Global dynamic memory.

**Case Thread which does not use default stack size (Object specific memory allocation)**

```
const osThreadAttr_t thread1_attr = {
  .stack_mem  = NULL
  .stack_size = 800
}
```
Using this definition, I have to do some changes in RTX_Config.c

- Increase Number of user threads with user-provided stack
- Increase Total Stack Size for threads with user-provided stack

**Case Thread which does not use default stack size (Object specific memory allocation)**

```
static uint64_t thread1_stk_1[64];
 
const osThreadAttr_t thread1_attr = {
  .stack_mem  = &thread1_stk_1[0];
  .stack_size = sizeof(thread1_stk_1);
}
```
Using this definition, I have to do some changes in RTX_Config.c

- Increase Number of user threads with user-provided stack
- **No** Increase Total Stack Size for threads with user-provided stack **is required**

**Additional I have some corrections or questions to Help**

Description
The Thread Management function group allows defining, creating, and controlling thread functions in the system. **The function main is a special thread function that is started at system initialization and has the initial priority osPriorityNormal.**

-> Main is no longer started as thread

Description of struct osThreadAttr_t attr_bits in the last version default value is  osThreadDettached not osThreadJoinable

**Message Queue**

Message queue is capable to transfer data objects of any size. So figure is not valid.
You have added an example using memory pools. For small objects I assume it is better to transport the objects without using memory pools. For large objects it is better to use memory pool to avoid the two copy actions. 

In your example (osMessageQueuePut()), you are using timeout values NULL. I would prefer to use 0 for this use.

You are using timeout values 0 (NULL) within the threads. So the threads are always in state ready or running. Only the use of osThreadYield() or end of round robin time leads to a thread change. You could use osWaitForever instead.
`osStatus_t osMessageQueuePut (osMessageQueueId_t mq_id, const void *msg_ptr, uint8_t msg_prio, uint32_t millisec)`
timeout parameter is still millisec not timeout

MatthiasHertel80, 2016-10-28:
Hello Ferdinand,
thanks for your valuable feedback. 
_"Case Thread which does not use default stack size_" is a very good idea for an example. I will add these examples to the documentation including necessary changes to the configuration. 
_"Main is no longer started as thread"_ / _osThreadJoinable no more default_ - Consider this solved.
_"Message Queue"_  This example will be reworked. The more complex example using a Memory Pool will be added as template variant.

Have a nice weekend,
Matthias

eugene345, 2016-11-14:
I think, that MsgQueue.c template has a mistake here 

` mid_MsgQueue = osMessageQueueNew(MSGQUEUE_OBJECTS, sizeof(MSGQUEUE_OBJ_t), NULL);`

Corrected:

` mid_MsgQueue = osMessageQueueNew(MSGQUEUE_OBJECTS, sizeof(MSGQUEUE_OBJ_t *), NULL);`

MatthiasHertel80, 2016-11-15:
osMessageQueueId_t osMessageQueueNew (uint32_t msg_count, **uint32_t msg_size**, const osMessageQueueAttr_t *attr)

The second parameter is the actual size of a message object. This is required to allocate the proper queue size (msg_count * msg_size).

eugene345, 2016-11-15:
Yes, it is. But, I thought, that this template uses memory pool to store messages, and message queue to pass pointer to a block in the memory pool. So message queue operates with just pointers to data. Is it true?

MatthiasHertel80, 2016-11-15:
You are right. I was looking at the wrong line. Will be fixed. Thanks.

FerdinandEn, 2016-11-25:
The example foe message queue included in RTOS documentation is not correct. Attached I have corrected version of the example. Modified old lines are marked with // FE

`#include "cmsis_os2.h"                                           // CMSIS RTOS header file

/*----------------------------------------------------------------------------
 *      Message Queue creation & usage
 *---------------------------------------------------------------------------*/
  
void Thread_MsgQueue1 (void *argument);                   // thread function 1
void Thread_MsgQueue2 (void *argument);                   // thread function 2
osThreadId_t tid_Thread_MsgQueue1;                                // thread id 1
osThreadId_t tid_Thread_MsgQueue2;                                // thread id 2
 
#define MSGQUEUE_OBJECTS      16                                // number of Message Queue Objects
 
typedef struct {                                                // object data type
  uint8_t Buf[32];
  uint8_t Idx;
} MSGQUEUE_OBJ_t;
 
// FE osMemoryPoolId_t mpid_MemPool2;                                         // memory pool id
osMessageQueueId_t mid_MsgQueue;                                      // message queue id
 
int Init_MsgQueue (void) {
  
  mid_MsgQueue = osMessageQueueNew(MSGQUEUE_OBJECTS, sizeof(MSGQUEUE_OBJ_t), NULL);
  if (!mid_MsgQueue) {
    ; // Message Queue object not created, handle failure
  }
  
  tid_Thread_MsgQueue1 = osThreadNew (Thread_MsgQueue1, NULL, NULL);
  if (!tid_Thread_MsgQueue1) return(-1);
  tid_Thread_MsgQueue2 = osThreadNew (Thread_MsgQueue2, NULL, NULL);
  if (!tid_Thread_MsgQueue2) return(-1);
  
  return(0);
}
 
void Thread_MsgQueue1 (void *argument) {
// FE  MSGQUEUE_OBJ_t pMsg = 0;
  MSGQUEUE_OBJ_t txMsg;
  while (1) {
    ; // Insert thread code here...
//  FE  pMsg->Buf[0] = 0x55;                                           // do some work...
//  FE   pMsg->Idx    = 0;
    txMsg.Buf[0] = 0x55;                                           // do some work...
    txMsg.Idx    = 0;    
// FE   osMessageQueuePut (mid_MsgQueue, &pMsg, NULL, NULL);
    osMessageQueuePut (mid_MsgQueue, &txMsg, 0, NULL);    // FE Meaning of priority?????
    osThreadYield ();                                             // suspend thread
  }
}
 
void Thread_MsgQueue2 (void *argument) {

// FE MSGQUEUE_OBJ_t pMsg = 0;
  MSGQUEUE_OBJ_t rxMsg;
  osStatus_t status; // FE

  while (1) {
    ; // Insert thread code here...
// FE   status = osMessageQueueGet (mid_MsgQueue, &pMsg, NULL, NULL);     // wait for message
    status = osMessageQueueGet (mid_MsgQueue, &rxMsg, NULL, NULL);     // wait for message
    if (status == osOK) {
        ; // process data
    }
  }
}`

FerdinandEn, 2016-11-25:
The behavior of parameter msg_prio is not explained in documentation. What is better priority lower or higher number?

RobertRostohar, 2016-11-25:
Higher number represents higher priority - will be added to the documentation.

FerdinandEn, 2016-11-30:
You have updated the message queue documentation. The big example includes still code with errors. 
E. g. `MSGQUEUE_OBJ_t pMsg = 0;` or `pMsg->Buf[0] = 0x55;` instead of  `pMsg.Buf[0] = 0x55;`
See my post above.

JonatanAntoni, 2017-04-03:
Hi @FerdinandEn, the documentation issues should have been fixed in the meanwhile. May I ask you to take a look to the latest version. Please let us know if you have further remarks. Otherwise I would like to ask you to close this issue.

FerdinandEn, 2017-04-06:
Hi Antoni,
the example has still errors. Find attached a correct Version of the example.

`#include "cmsis_os2.h"                                             // CMSIS RTOS header file
 
/*----------------------------------------------------------------------------
 *      Message Queue creation & usage
 *---------------------------------------------------------------------------*/
  
void Thread_MsgQueue1 (void *argument);                            // thread function 1
void Thread_MsgQueue2 (void *argument);                            // thread function 2
osThreadId_t tid_Thread_MsgQueue1;                                 // thread id 1
osThreadId_t tid_Thread_MsgQueue2;                                 // thread id 2
 
#define MSGQUEUE_OBJECTS      16                                   // number of Message Queue Objects
 
typedef struct {                                                   // object data type
  uint8_t Buf[32];
  uint8_t Idx;
} MSGQUEUE_OBJ_t;
 
osMemoryPoolId_t mpid_MemPool2;                                    // memory pool id
osMessageQueueId_t mid_MsgQueue;                                   // message queue id
 
int Init_MsgQueue (void) {
  
  mid_MsgQueue = osMessageQueueNew(MSGQUEUE_OBJECTS, sizeof(MSGQUEUE_OBJ_t), NULL);
  if (!mid_MsgQueue) {
    ; // Message Queue object not created, handle failure
  }
  
  tid_Thread_MsgQueue1 = osThreadNew (Thread_MsgQueue1, NULL, NULL);
  if (!tid_Thread_MsgQueue1) return(-1);
  tid_Thread_MsgQueue2 = osThreadNew (Thread_MsgQueue2, NULL, NULL);
  if (!tid_Thread_MsgQueue2) return(-1);
  
  return(0);
}
 
void Thread_MsgQueue1 (void *argument) {
  MSGQUEUE_OBJ_t msg;

  while (1) {
    ; // Insert thread code here...
    msg.Buf[0] = 0x55;                                           // do some work...
    msg.Idx    = 0;
    osMessageQueuePut (mid_MsgQueue, &msg, 0, NULL);
    osThreadYield ();                                              // suspend thread
  }
}
 
void Thread_MsgQueue2 (void *argument) {

  MSGQUEUE_OBJ_t msg;
  osStatus_t status;

  while (1) {
    ; // Insert thread code here...
    status = osMessageQueueGet (mid_MsgQueue, &msg, NULL, NULL);  // wait for message
    if (status == osOK) {
        ; // process data
    }
  }
}`

JonatanAntoni, 2017-04-06:
Thank you for the feedback, @FerdinandEn. I am going to look into this. Or are you willing to provide this change as a pull request?

FerdinandEn, 2017-04-06:
[cmsis_os2_MessageNew.txt](https://github.com/ARM-software/CMSIS_5/files/902281/cmsis_os2_MessageNew.txt)

Find attached the changed file

JonatanAntoni, 2017-04-06:
Thank you for your contribution.
I applied your patch with [commit c1b423](https://github.com/ARM-software/CMSIS_5/commit/c1b4232f4a98275dd79ebc99cf9b5da5053def4d).

closed 2017-04-10 by FerdinandEn