ARM-software/CMSIS_5 #96: Task state hooks in user application

offchooffcho, 2016-10-27:
Hello,

This is based on experience w/ CMSIS-RTOS RTX part of CMSIS 4.X, but I can imagine it is still applicable for CMSIS 5.

There is a basic mechanism for tasks profiling (in file SRC\HAL_CM.c we have the dbg_init, dbg_task_notify and dbg_task_switch). It relies on ITM and user cannot modify it in any way.

For applications in which trace & analyze are not just "good to have", but require more complex tooling, these are quite limited. So people fall back to modifying the RTOS, build it with a different version of the compiler (for instance - the IAR build delivered by ARM is based on IAR EW 6.4 and project teams last years work with 7.X). Consequently they need to validate it again and considering how many people actually understand what RTOS does internally and how to test it, this places big risk onto whoever tries to do such thing.

So, I would like to ask about the possibilities to have user defined hooks support in the standard RTOS deliverable.  Having the existing hooks "weak" and the user can choose if to override them or not seems like simple and straightforward solution.

RobertRostohar, 2016-11-28:
RTX5 has been extended with hooks that can be used to trace its functionality (API, operations, errors). This is not limited to threads but is available also for all other objects (semaphore, mutex, ...).

Hooks are routed by default to the Event Recorder component but can be overridden by user.

ReinhardKeil, 2017-02-27:
Closing this issue as there is no further activity.  Note that this is part of the CMSIS 5.0.1 release.

closed 2017-02-27 by ReinhardKeil