ARM-software/CMSIS_5 #98: Error when compiling with GCC6

clementperon, 2016-10-27:
I tried to compile the CMSIS 6.0.0 Beta 4 DSP for CM4F using gcc6 but i got this error. Everything is fine using gcc5

```c
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c: In function 'arm_radix4_butterfly_q31':
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:678:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xaya = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:683:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xbyb = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:688:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xcyc = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:693:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xdyd = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c: In function 'arm_radix4_butterfly_inverse_q31':
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:1318:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xaya = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:1323:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xbyb = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:1328:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xcyc = *__SIMD64(ptr1)++;
     ^~~~
~/cmsis/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c:1333:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     xdyd = *__SIMD64(ptr1)++;
```

clementperon, 2016-11-03:
Declare ptr1 as a void pointer fix the issue and cast to int32 or int64.

```c
---------- CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q31.c ----------
index 0fbb40b..e9e751e 100644
@@ -413,7 +413,7 @@ void arm_radix4_butterfly_q31(
   q31_t xa_out, xb_out, xc_out, xd_out;
   q31_t ya_out, yb_out, yc_out, yd_out;
 
-  q31_t *ptr1;
+  void *ptr1;
   q31_t *pSi0;
   q31_t *pSi1;
   q31_t *pSi2;
@@ -726,32 +726,32 @@ void arm_radix4_butterfly_q31(
     ya_out = ya + yb + yc + yd;
 
     /* pointer updation for writing */
-    ptr1 = ptr1 - 8u;
+    __SIMD32(ptr1) -= 8u;
 
     /* writing xa' and ya' */
-    *ptr1++ = xa_out;
-    *ptr1++ = ya_out;
+    *__SIMD32(ptr1)++ = xa_out;
+    *__SIMD32(ptr1)++ = ya_out;
 
     xc_out = (xa - xb + xc - xd);
     yc_out = (ya - yb + yc - yd);
 
     /* writing xc' and yc' */
-    *ptr1++ = xc_out;
-    *ptr1++ = yc_out;
+    *__SIMD32(ptr1)++ = xc_out;
+    *__SIMD32(ptr1)++ = yc_out;
 
     xb_out = (xa + yb - xc - yd);
     yb_out = (ya - xb - yc + xd);
 
     /* writing xb' and yb' */
-    *ptr1++ = xb_out;
-    *ptr1++ = yb_out;
+    *__SIMD32(ptr1)++ = xb_out;
+    *__SIMD32(ptr1)++ = yb_out;
 
     xd_out = (xa - yb - xc + yd);
     yd_out = (ya + xb - yc - xd);
 
     /* writing xd' and yd' */
-    *ptr1++ = xd_out;
-    *ptr1++ = yd_out;
+    *__SIMD32(ptr1)++ = xd_out;
+    *__SIMD32(ptr1)++ = yd_out;
 
 
   } while(--j);
@@ -830,7 +830,7 @@ void arm_radix4_butterfly_inverse_q31(
   q31_t xa_out, xb_out, xc_out, xd_out;
   q31_t ya_out, yb_out, yc_out, yd_out;
 
-  q31_t *ptr1;
+  void *ptr1;
   q63_t xaya, xbyb, xcyc, xdyd;
 
   /* input is be 1.31(q31) format for all FFT sizes */
@@ -1061,7 +1061,7 @@ void arm_radix4_butterfly_inverse_q31(
   q31_t xa_out, xb_out, xc_out, xd_out;
   q31_t ya_out, yb_out, yc_out, yd_out;
 
-  q31_t *ptr1;
+  void *ptr1;
   q31_t *pSi0;
   q31_t *pSi1;
   q31_t *pSi2;
@@ -1366,32 +1366,32 @@ void arm_radix4_butterfly_inverse_q31(
     ya_out = ya + yb + yc + yd;
 
     /* pointer updation for writing */
-    ptr1 = ptr1 - 8u;
+    __SIMD32(ptr1) -= 8u;
 
     /* writing xa' and ya' */
-    *ptr1++ = xa_out;
-    *ptr1++ = ya_out;
+    *__SIMD32(ptr1)++ = xa_out;
+    *__SIMD32(ptr1)++  = ya_out;
 
     xc_out = (xa - xb + xc - xd);
     yc_out = (ya - yb + yc - yd);
 
     /* writing xc' and yc' */
-    *ptr1++ = xc_out;
-    *ptr1++ = yc_out;
+    *__SIMD32(ptr1)++  = xc_out;
+    *__SIMD32(ptr1)++  = yc_out;
 
     xb_out = (xa - yb - xc + yd);
     yb_out = (ya + xb - yc - xd);
 
     /* writing xb' and yb' */
-    *ptr1++ = xb_out;
-    *ptr1++ = yb_out;
+    *__SIMD32(ptr1)++  = xb_out;
+    *__SIMD32(ptr1)++  = yb_out;
 
     xd_out = (xa + yb - xc - yd);
     yd_out = (ya - xb - yc + xd);
 
     /* writing xd' and yd' */
-    *ptr1++ = xd_out;
-    *ptr1++ = yd_out;
+    *__SIMD32(ptr1)++ = xd_out;
+    *__SIMD32(ptr1)++ = yd_out;
 
   } while(--j);
 ```

llefaucheur, 2018-04-11:
Hi Clement,
Could the issue be related to the Beta version ?  I am sorry I cannot reproduce the issue with the closed t gcc compiler I could have access to (v6.3.1) : 

IDE-Version: µVision V5.24.2.0
Toolchain:       MDK-ARM Professional  Version: 5.24.1
Toolchain Path:  C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin 
C Compiler:      GCC.exe       VERSION 6.3.1

Rebuild target 'cortexM4lf'
compiling arm_abs_q15.c...
compiling arm_abs_f32.c...
compiling arm_abs_q7.c...
.. 
compiling arm_cfft_radix4_init_q31.c...
compiling arm_cfft_radix4_q15.c...
compiling arm_cfft_radix4_q31.c...
compiling arm_dct4_f32.c...
..
".\IntermediateFiles\cortexM4lf\libarm_cortexM4lf_math.a" - 0 Error(s), 0 Warning(s).

clementperon, 2018-04-11:
Hi Laurent,

Can't reproduce it anymore, closing this issue.

Clement

closed 2018-04-11 by clementperon

Summercore, 2018-12-04:
Hi, all

I am a software engineer from NXP and want to know which branch can get this patch in https://github.com/ARM-software/CMSIS_5, we will fetch it into our SDK. Thanks a lot.

Best Regards,
Summer

JonatanAntoni, 2018-12-04:
Hi @Summercore,

this issue has not been fixed explicitly. The recent release of CMSIS should be the best for productive use. Please consider referencing our released pack instead of duplicating CMSIS files into your code base. Referencing the pack gives the users direct access to all bug fixes we add in future releases.

Cheers,
Jonatan

Summercore, 2018-12-04:
Hi, @JonatanAntoni 

Thank for you reply and I mean that we clone your CMSIS package directly into our SDK, so now use your launched package and waiting for it fixed explicitly.

Regards,
Summer

JonatanAntoni, 2018-12-04:
Hi @Summercore,

this issue has been closed and there will be no explicit fix. What exactly are you looking for?

Do you still see issues with the latest release? If yes, please report them in a new issue. You might reference to this one if you think its related.

Cheers,
Jonatan

Summercore, 2018-12-05:
Hi @JonatanAntoni 

Thank for your reply.
I have test it and already have been fixed.

Regards,
Summer