ARM-software/CMSIS_5 #99: RTOS2 Debug

eugene345, 2016-10-27:
I found that new feature was added recently. But after I choose Debug variant for Keil RTX5 component in   the "manage run-time environment" window I see that new component required, which is not available from current packs:

[image]

And another question. Do you plan to integrate CMSIS 5 with the "Event viewer" and "System and thread viewer"? I use "Component viewer" now, but it is not so informative as tools above.

RobertRostohar, 2016-10-28:
The required dependency on Event Recorder is removed now (it will be optional to use it - not yet supported).

"Component Viewer" is a replacement for "System and thread viewer" and provides much more information.

"Event Viewer" is being replaced with "Event Recorder" which will also provide more events (not just thread switches).

eugene345, 2016-11-14:
Is there any option to show in the Component viewer information about dynamically allocated objects?

I can see the new component - Event recorder. But it uses CMSIS-DAP.  Does it mean that high-grade debuggers (such as ULINKpro) become useless when debugging RTOS applications? Because there is no CMSIS-DAP driver for it. 
And also CMSIS-DAP doesn't support ETM. So, if driver appears, adapter will loose access to ETM.

RobertRostohar, 2016-11-15:
> Is there any option to show in the Component viewer information about dynamically allocated objects?

Yes, dynamically allocated objects should be shown.

> Event recorder. But it uses CMSIS-DAP

It does not use CMSIS-DAP but Debug Access Port (DAP) and will work with any debugger connected through DAP (JTAG/SWD). There is no need for any CMSIS-DAP driver or any other adaptation of existing debuggers..

eugene345, 2016-11-15:
Thanks for the explanation.
Does RTOS2 have event annotations to work with Event recorder now?

jankap, 2016-11-29:
> Thanks for the explanation.
> Does RTOS2 have event annotations to work with Event recorder now?

That is interesting for me too, because with uvision 5.22, RTX5 and CMSIS-RTOS2 the event recorder remains empty (except my own test messages) and the event viewer (which is very important for us, because of the good overview in time) does not display the threads, only exceptions. The latter makes sense because I couldn't find any source code writing to ITM Port 31 anymore (HAL_CM.c has done this before). Is there a timeline when these features are expected to work again?

Thank you!

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil