ARM-software/CMSIS_5 #101: CMSIS RTOSv2 API:

ilg-ul, 2016-10-28:


ReinhardKeil, 2016-11-15:
An example for using this functions is provided, take a look here http://arm-software.github.io/CMSIS_5/RTOS2/html/TickLess.html.

ilg-ul, 2016-11-15:
again, this seems a RTX specific functionality. is it mandatory for all CMSIS RTOSv2 implementations?

to me this seems quite unfortunate and limitative, since it messes functionality from the scheduler with functionality from the ticks timer.

CMSIS++ (and POSIX in general) defines multiple timers (systick, rtc, high_res, etc), and it is possible to enter either a shallow sleep, a deep sleep or any custom sleep, based on **any** timer.

if this is a deep sleep, the application should figure out a method to compute how much time passed while sleeping, and update **all** clocks, so your RTX specific functions are not enough in the general case.

my suggestion is to keep these function RTX specific, and not make them mandatory for CMSIS RTOS.

ReinhardKeil, 2017-02-27:
Note that the documentation has moved to: http://arm-software.github.io/CMSIS_5/RTOS2/html/theory_of_operation.html#TickLess

You are correct, this is apparently RTX specific.  We will investigate into a generic solution as we are working currently on a FreeRTOS adoption of CMSIS-RTOS v5.  See also here: https://github.com/ARM-software/CMSIS-FreeRTOS

JonatanAntoni, 2019-07-30:
The functions are well documented here: [osKernelSuspend](https://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__KernelCtrl.html#gae26683e1606ec633354a2876c68f0c1f). Does this solve this issue?

ReinhardKeil, 2019-10-24:
Closing as there is no further activity - initial question is resolved.

closed 2019-10-24 by ReinhardKeil