ARM-software/CMSIS_5 #102: CMSIS RTOSv2 API: osKernelLock() / osKernelUnlock() not nesting safe

ilg-ul, 2016-10-28:
If these two functions are your solution to implementing scheduler critical sections, then not only the names are not very inspired (they do not lock the kernel, but the scheduler), but the API is not nesting safe.

The point with these calls is that normally when entering a critical section it is not known if the scheduler is currently locked or not; regardless of this, inside the critical section the scheduler must be locked, and when the critical section is exited, the scheduler lock status must be exactly as before entering the scheduler section.

The complete API, as defined by [CMSIS++](http://micro-os-plus.github.io), is the following:

```
os_sched_state_t 	os_sched_lock (void);
os_sched_state_t 	os_sched_unlock (void);
os_sched_state_t 	os_sched_set_locked (os_sched_state_t state);
```

With these functions, implementing a scheduler critical section looks like this:

```
void f(void)
{
  // ... 
  {
    // ===== Enter scheduler critical section ===== 
    os_sched_state_t s = os_sched_lock();
    // ... no context switches can occur here ...
    os_sched_set_locked(s);
    // ===== Exit scheduler critical section ===== 
  }
  // ...
}
```

Nested critical sections occur when in a critical section like before is called another function, which includes its own critical section; using lock/unlock instead of save/restore will prematurely exit the critical section when the inner unlock is executed.

The `os_sched_unlock()` function is not the pair of `os_sched_lock()`, but the complement of it, used to define _uncritical sections_, which are regions of code where the context switches are allowed.

```
void f(void)
{
  // ... 
  {
    // ===== Enter scheduler critical section ===== 
    os_sched_state_t sc = os_sched_lock();
    // ... no context switches can occur here ...
      {
        // ===== Enter scheduler uncritical section ===== 
        os_sched_state_t su = os_sched_unlock();
        // ... context switches can occur here, for example you can yield() ...
        os_sched_set_locked(su);
        // ===== Exit scheduler uncritical section ===== 
      }
    // ... no context switches can occur here ...
    os_sched_set_locked(sc);
    // ===== Exit scheduler critical section ===== 
  }
  // ...
}
```

ilg-ul, 2016-11-14:
@RobertRostohar, did you analyse this issue?

Not being able to nest critical sections is a serious design problem, that usually leads to shorter than needed critical sections, and consequently to synchronisation problems.

RobertRostohar, 2016-11-17:
The following shows typical usage (nesting safe):

```
{
  uint32_t lock = osKernelLock();
  // ...
  if (lock) {
    osKernelUnlock();
  }
}
```

ilg-ul, 2016-11-17:
this solution is not symmetrical, you cannot implement un-critical sections, i.e. sections where, regardless what the scheduler status is, you temporarily unlock and then return to initial status.

ilg-ul, 2016-11-17:
and, how can you prevent users taking the shortcut, without checking the return value?

or, more generally, do you think this API would pass Steve Meyers' criteria:

> Make Interfaces Easy to Use Correctly and Hard to Use Incorrectly

ReinhardKeil, 2016-11-17:
Sorry I don't get this problem

*this solution is not symmetrical, you cannot implement un-critical sections, i.e. sections where, regardless what the scheduler status is, you temporarily unlock and then return to initial status.*

@ilg-ul can you explain why it is impossible to return to the initial status using this code:

{
  uint32_t lock = osKernelLock();
  // ...
  if (lock) {
    osKernelUnlock();
  }
}

ilg-ul, 2016-11-17:
your code returns to the initial status only after a lock(), you cannot write something like

```
{
  uint32_t lock = osKernelUnlock();
  // ...
  if (lock) {
    osKernelLock();
  }
}
```

also the idea to mandate the user to conditionally issue calls in a certain order is a weakness.

the theory behind critical sections is simple, regardless the current status, the critical section is entered with a function that prevents some concurrent access, and the critical section is exited by restoring the initial status, regardless what it is.

exactly in the mirror, un-critical sections are entered with a function that permits concurrent access, and exits by restoring the initial status, regardless what it is.

thus, the complete API has exactly three functions, lock(), unlock(), restore(), used in pairs, lock()+restore(), unlock()+restore(); both lock() and unlock() return a status, used by restore() as parameter. (the example in the initial post should have been clear enough).

with these three functions you can implement any critical and un-critical sections, the usage is symmetrical and the interaction with the user minimal, the user needs to preserve the value obtained when entering the section and use it to exit the section.

RobertRostohar, 2016-11-25:
API has been updated to support critical and uncritical sections (lock, unlock, restore) as suggested.

ilg-ul, 2016-11-25:
great!

ilg-ul, 2016-11-25:
once you fixed the scheduler critical sections, you should address interrupt critical sections, using a similar approach. interrupts critical sections are similar, just that they do not lock the scheduler, but the interrupts.

there was an old discussion at https://github.com/ARM-software/CMSIS/issues/38, the main ideas are there, but probably some updates are required.

as implementation, in CMSIS++ it is possible to decide if the interrupts are totally disabled, or BASEPRI is used to set a certain level. interrupts with higher priorities than BASEPRI (lower values) are not allowed to use system calls. simple cores, that do not have BASEPRI, do not have this option.

the current Cortex-M implementation is available from https://github.com/micro-os-plus/micro-os-plus-iii-cortexm/blob/xpack/include/cmsis-plus/rtos/port/os-inlines.h

ReinhardKeil, 2017-02-14:
completed

closed 2017-02-14 by ReinhardKeil