ARM-software/CMSIS_5 #103: CMSIS build error in IAR

TarasMarkiv, 2016-11-05:
Hello,
I have a problem when build the project with CMSIS 5.0 in IAR IDE. Part of the build log is below:

Error[Pe020]: identifier "__INLINE" is undefined C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3\arm\inc\c\cmsis_iar.h 211 
Error[Pe101]: "uint32_t" has already been declared in the current scope (at line 27 of "C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3\arm\inc\c\stdint.h") C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3\arm\inc\c\cmsis_iar.h 211 
Error[Pe065]: expected a ";" C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3\arm\inc\c\cmsis_iar.h 211 
Warning[Pe012]: parsing restarts here after previous syntax error d:\path\\cmsis\include\cmsis_compiler.h 76 
Warning[Pe1105]: #warning directive: No compiler specific solution for __ALIGNED. __ALIGNED is ignored. d:\path\\cmsis\include\cmsis_compiler.h 80 
Error while running C/C++ Compiler 

The cmsis_compiler.h file includes a <cmsis_iar.h> file before __STATIC_INLINE macro and other are defined(line 55).

I would apreciate for a quick fix.

Thanks,
Taras

ReinhardKeil, 2016-11-08:
We did report this problem to IAR.  IAR should support the new macros in cmsis_iar.h.  This seems to be not the case yet (or you are using an old IAR version).

GuentherMartin, 2016-11-09:
Problem should be fixed with latest version of cmsis_compiler.h.
Please see https://github.com/ARM-software/CMSIS_5/commit/c0083dddd7ce6aaee51a8ef8c12de18218ea0f61

TarasMarkiv, 2016-11-09:
Thanks a lot for fixing this issue.

closed 2016-11-09 by TarasMarkiv