ARM-software/CMSIS_5 #104: CMSIS v5.0 warning for IAR

IuriiB, 2016-11-08:
cmsis_compiler.h file contains the following code snippet:
#ifndef __ALIGNED
    #warning No compiler specific solution for __ALIGNED. __ALIGNED is ignored.
    #define __ALIGNED(x)
#endif

Which produce build warning for each project in IAR. 
Please add implementation for __ALIGNED(x) and remove the warning generation.
Implementation could be done with using pragma directive.

GuentherMartin, 2016-11-09:
Please see https://github.com/ARM-software/CMSIS_5/issues/103

IuriiB, 2016-11-11:
Hi Martin,

I can see that case #103 was fixed, but __ALIGNED(x) implementation was not added for IAR compiler.
Maybe I missed something, but how can I verify the fix of this case (#104)?

Thank you,
Iurii

GuentherMartin, 2016-11-11:
Hello IuriiB,

please excuse the missleading comment.

Currently we do not have a solution for macro __ALIGNED(x) for IAR compiler.
The warning is just a hint for the user that this macro is ignored.

It is possible to modify cmsis_compiler.h file for your needs so you can remove the warning if the macro is not used.

David-Garcia-Polo, 2016-11-11:
You can try the following

#ifndef __ALIGNED
        #define STRING_PRAGMA(x) _Pragma(#x)
        #define __ALIGNED(n) STRING_PRAGMA(data_alignment = n)
#endif

IuriiB, 2016-11-11:
Hi David,
Thank you for the feedback. Actually we use this approach as you suggested in post above. But I think this should go as part of SMSIS, because it provides __ALIGNED implementation for GCC and MDK compiler. So it looks more reasonable for me during creation cross compiler code use SMSIS defines instead of using some redefines, specific for IAR.
Thank you,
Iurii

closed 2016-11-11 by ReinhardKeil