ARM-software/CMSIS_5 #106: rtx_config.c

joabeck, 2016-11-13:
I have an issue with the RTOS config file rtx_config.c. This file contains the #defines which configure the different Rtos object counts (for example OS_MUTEX_NUM etc.).
Sometimes it would be helpful to have this constants available outside Rtos (for example for a C++  wrapper around Timers, Threads etc).
But this is not possible, because they are defined inside a .c file and not as usual for #define constants inside a .h file.
I think it would be a good idea to separate the rtx_config. c into a .h and a .c file.

What do you think about this?

Regards
Joachim

RobertRostohar, 2016-11-14:
rtx_config.c is specific to RTX implementation of CMSIS RTOS2. Therefore application code should not rely on its definitions in order to be portable. 

Also the mentioned object counts are relevant only if enabled which would make it even harder to use. It is not intended that application code would use those defines.

ReinhardKeil, 2016-11-15:
To get it right, I would like to understand a bit more about the use-case and how we should structure the header files. At the end, it is today possible to modify the rtx_config.c file and separating this file is also not a big task. Would this be already sufficient?

joabeck, 2016-11-16:
In our Projects it is not allowed to use dynamic Memory (due to possible memory fragmentation). Because of this reason we want to use RTX2 with fixed size memory blocks for every object. We are using a C++ wrapper to connect RTX to the C++ world. In this wrapper we need some static tables which must have the same size as the maximum count of RTX objects (Task Count, Timer Count etc.).
To avoid doubling the definition, it would be nice to have access to the defines in rtx_config.c.

Yes, separating defines and code would be sufficient for us.
Best regards
Joachim

ilg-ul, 2016-11-16:
@joabeck, FYI, in [CMSIS++](http://micro-os-plus.github.io) by design all system primitives are C++ objects and can be statically allocated, if needed. and the native [RTOS API is C++](http://micro-os-plus.github.io/reference/cmsis-plus/group__cmsis-plus-rtos.html), you no longer need a C++ wrapper.

RobertRostohar, 2016-11-25:
RTX configuration file has been split into .h and .c as requested.

closed 2016-11-30 by joabeck