ARM-software/CMSIS_5 #107: RTOS2 using Selection Keil RTX5 Source

FerdinandEn, 2016-11-14:
Writing a program with RTOS2 using selection source, I got the error

`C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.0.0\CMSIS\RTOS2\RTX\Source\core_cm.h(307): error:  #136: struct "<unnamed>" has no field "SHP"`

in a lot of RTOS source functions.

Using selection Library no compile error and the program works fine.

Source is the Default selection for Keil RTX5

RobertRostohar, 2016-11-14:
Which device/core are you using?

Please check if you have an RTE_Components.h file with the appropriate definition.

Example:
`#define CMSIS_device_header "ARMCM3.h"
`

FerdinandEn, 2016-11-14:
I am using ARMCM7_SP or STM32F746ZGTx. In both cases the same errors. The line is not included. I have deleted RTE_Components.h. But the line is not included after regeneration.

RobertRostohar, 2016-11-14:
It seems that CMSIS Core header for Cortex-M7 defines SHPR (Cortex-M3/M4 defines SHP).

core_cm.h will be updated to handle this anomaly.

RobertRostohar, 2016-11-14:
In order for the mentioned device define to be generated in RTE_Components.h the latest uVision is required (recently released in MDK522).

RobertRostohar, 2016-11-16:
"SHP" issued for Cortex-M7 should be fixed now.

closed 2016-11-30 by FerdinandEn