ARM-software/CMSIS_5 #109: RTOS2 using Selection Keil RTX5 Source Expected an Expression

ablaylock, 2016-11-14:
Compiling the CMSIS-RTOS2 Blinky (uVision Simulation) example provided for simulation in the ARM pack in MDK 5.22 won't compile with the following output:

`
*** Using Compiler 'V5.06 update 4 (build 422)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Simulation'
assembling irq_cm3.s...
compiling rtx_kernel.c...
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.0.0\CMSIS\RTOS2\RTX\Source\rtx_kernel.c(31): error:  #29: expected an expression
  { .os_id = os_KernelId, .version = os_CMSIS_RTX, .kernel.state = os_KernelInactive };
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.0.0\CMSIS\RTOS2\RTX\Source\rtx_kernel.c(31): error:  #29: expected an expression
  { .os_id = 
s_KernelId, .version = os_CMSIS_RTX, .kernel.state = os_KernelInactive };
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.0.0\CMSIS\RTOS2\RTX\Source\rtx_kernel.c(31): error:  #29: expected an expression
  { .os_id = 
s_KernelId, .version = 
s_CMSIS_RTX, .kernel.state = os_KernelInactive };
C:\Keil_v5\ARM\PACK\ARM\CMSIS\5.0.0\CMSIS\RTOS2\RTX\Source\rtx_kernel.c: 0 warnings, 3 errors
compiling rtx_memory.c...
compiling rtx_delay.c...
compiling rtx_evflags.c...
compiling rtx_mempool.c...
compiling rtx_msgqueue.c...
compiling rtx_mutex.c...
compiling rtx_semaphore.c...
compiling retarget_io.c...
compiling rtx_system.c...
compiling rtx_timer.c...
compiling rtx_thread.c...
".\Flash\Blinky.axf" - 3 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:00
`

ablaylock, 2016-11-14:
It seems to be complaining about os_KernelId and

RobertRostohar, 2016-11-15:
Example is configured to use RTX5 library by default and should build without errors. When RTX5 sources are selected instead then C99 needs to be enabled in order to build.

ReinhardKeil, 2016-11-15:
C99 is selected under Project - Options - C/C++ - Language C

JonatanAntoni, 2017-02-27:
I recently added minor enhancements to the documentation/examples on branch issue-109.

JonatanAntoni, 2017-03-07:
The documentation change has been merged to develop.

closed 2017-03-07 by JonatanAntoni