ARM-software/CMSIS_5 #110: Interrupt may occur after calling NVIC_DisableIRQ

c1728p9, 2016-11-14:
The macro [NVIC_DisableIRQ](https://github.com/ARM-software/CMSIS_5/blob/4da2d09bb336e1c08ad54fa9988658eccaa8d169/CMSIS/Core/Include/core_cm4.h#L1695) is not safe to use for interrupt protection since interrupts can still occur for a short amount of time after this instruction. For reference the docs explaining this can be found [here](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0321a/BIHHFHJD.html). Should a DSB and ISB be added to this macro to guarantee interrupts will not fire after the call?

Original issue reported here: https://github.com/ARMmbed/mbed-os/issues/2242

GuentherMartin, 2016-11-17:
Function NVIC_DisableIRQ is updated.
Instructions dsb and isb are added after  write to interrupt enable register.
Please see https://github.com/ARM-software/CMSIS_5/commit/d1a283ace989eadf7e3298d1219de62fe40270d4

c1728p9, 2016-11-17:
Thanks for the quick fix @GuentherMartin. Closing this issue.

closed 2016-11-17 by c1728p9

kjsdf7wr3, 2021-10-13:
I think `disable_irq` and `set_PRIMASK` also need the same barriers as `NVIC_DisableIRQ`? Currently only `NVIC_DisableIRQ` has the barrier instructions (looking at cmsis_gcc.h).