ARM-software/CMSIS_5 #112: arm_math.h: macro TABLE_SIZE tend to cause name clashes

franzhollerer-Artesyn, 2016-11-16:
From arm_math.h:

```c
/**
   * @brief Macros required for SINE and COSINE Fast math approximations
   */

#define FAST_MATH_TABLE_SIZE  512
#define FAST_MATH_Q31_SHIFT   (32 - 10)
#define FAST_MATH_Q15_SHIFT   (16 - 10)
#define CONTROLLER_Q31_SHIFT  (32 - 9)
#define TABLE_SIZE  256
#define TABLE_SPACING_Q31     0x400000
#define TABLE_SPACING_Q15     0x80
```

The very generic macro *TABLE_SIZE* in a public interface tend to cause name clashes. I suggest to give it a more specific name.

GuentherMartin, 2016-12-23:
Obsolete macro TABLE_SIZE is removed from arm_math.h.
Please see https://github.com/ARM-software/CMSIS_5/commit/cec7ea463b580e2fca24be5ff7d18fbfacc336f4

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil