ARM-software/CMSIS_5 #113: Event Recorder with RTOS2 - how to use

FerdinandEn, 2016-11-17:
How can I use Event Recorder with RTOS2?

I have changed the setting  to Source.

ReinhardKeil, 2016-11-17:
The team is currently adding the stubs for the Event Recorder to RTOS2.  This should be finished in about 2 weeks.

RobertRostohar, 2016-11-25:
Support for Event Recorder has been added to RTX5.

ReinhardKeil, 2017-02-27:
This is no part of CMSIS 5.0.1. Therefore closing this issue.

closed 2017-02-27 by ReinhardKeil

rizickus, 2017-05-19:
I bet it is not the right place to ask, but...
Is there any way how to save Event Recorder information to file using command line or debug script, instead of clicking on IDE? We would like to automate some parts of our debug routines. Event Recorder is really helpful tool and ability to automatically collect events directly to csv file would be very useful.

JonatanAntoni, 2017-05-23:
Hi @rizickus.
The Event Recorder integration into MDK/uVision is not part of CMSIS.
The tools team always has command line support on the backlog. Please consider to file a customer service request at [silver.arm.com](http://www2.keil.com/support/silver) if you have any tool issue.