ARM-software/CMSIS_5 #115: Absent necessary definitions for the Core Debug Registers in core_cm0.h file

IuriiB, 2016-11-20:
core_cm4.h file contains necessary definitions for the DHCSR (Debug Halting Control and Status Register) register. But core_cm0.h file doesn't.
There is the following comment in core_cm0.h file:
  "\defgroup CMSIS_CoreDebug Core Debug Registers (CoreDebug)
  \brief Cortex-M0 Core Debug Registers (DCB registers, SHCSR, and DFSR) are only accessible over DAP and not via processor. Therefore they are not covered by the Cortex-M0 header file."

However per ARM info center DHCSR should be available for the CM0+ core. (see http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0774f/chr1359124993371.html or http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/CEGCJAHJ.html).

I am trying to examine the DHCSR register (for both CM4 and CM0+ cores) per example from http://www.keil.com/appnotes/files/apnt209.pdf (code snippet for "Implementing Fault Handlers" page 10).

Could you please help and clarify what is true.

GuentherMartin, 2016-11-25:
Hello Iurii,

in ARMv6-M Architecture Reference Manual chapter C1.6 Debug register support in the SCS is a note saying:
"The DWT, BPU, ROM table, DCB, and the SHCSR and DFSR registers are accessible through the DAP
interface. Access from the processor is IMPLEMENTATION DEFINED."

The DCB register (e.g. DHCSR) are not implemented on a Cortex-M0, Cortex-M0+.
They behave as "RAZ".

I have tested this on ARM's Cortex-M Prototyping System.

Best Regards, Martin

IuriiB, 2016-11-29:
Hello Martin,

Thank you for the feedback. 
Could you please advise me what approach could be used instead of checking the DHCSR register to test if CM0+ core is in debug mode or not in firmware?

Best Regards, Iurii

GuentherMartin, 2016-11-30:
Hello Iurii,

sorry to say but I can not give you an advice.
It is difficult on a Cortex-M0 or Cortex-M0+ to determine if the code runs under control of a debugger or not.
I have found the following forum entry http://www.keil.com/forum/60196/ where the user tries as workaround the use of a breakpoint instruction to detremine debug mode or not.

Best Regards, Martin

IuriiB, 2016-12-05:
Hello Martin,

Thank you for the help.

Best Regards, Iurii

closed 2016-12-05 by IuriiB