ARM-software/CMSIS_5 #116: Problem with the configuration of thread stacks

joabeck, 2016-11-23:
Hello,

i have had the following support case in ARM support. Andreas Barth from ARM suggests to open an issue here about the problem. The easiest way to explain the problem is to quote the support case:

*** 2016-11-19  16:24 UTC *** Beck, Joachim/Unknown Company:1366868 j.beck@kbs-gmbh.de ***

Dear Sirs,

i have some trouble with stack configuration of threads in CMSIS 5 / RTOS2.
I want to use only external supplied stacks (via os_Thread_Attr). I cannot figure out how to set the different defines in rtx_config.c. Every combination i try either leads to an error message or to a reservation of thread memory (for user threads) inside the RTOS which I do not need (i know i need RTOS internal stack for the timer and the idle thread).
What is the right configuration when I want to supply an external stack for all user threads without wasting memory in the RTOS configuration. 
Thank you for your support.
Best regards
Joachim

------------------------------------------------------------------------------------------------------
Public text added by Barth, Andreas/ARM:1187425 at 2016-11-22 11:26:55
Hello Joachim,
not sure what the problem is. Maybe you need to send us an example that demonstrates your issue.

I just tried in a simple example:

uint32_t Thread_1_Stack[ 0x1000 / sizeof( uint32_t ) ] ;

int main( void )
{
  osThreadAttr_t ThreadAttr = { 0 } ;

  ThreadAttr.stack_mem = Thread_1_Stack ;
  ThreadAttr.stack_size = sizeof( Thread_1_Stack ) ;
  osThreadNew( Thread_1, NULL, &ThreadAttr ) ;    // Create application main thread

And when inside Thread_1, the stack pointer is in range of the Thread_1_Stack object.

Best regards,
Andreas Barth
Public text added by Beck, Joachim/Unknown Company:1366868 at 2016-11-22 15:38:44
This is not my problem, your example works for me too.
 
------------------------------------------------------------------------------------------------------
The problem is, that the RTOS reserves (wastes) stack space even if i supply the stack via the attributes from external (as in your example). And i found no way to avoid this effect. If i interpret the map file correctly, RTOS always reserves a default sized stack internal which is not used for anything.    

Best regards
Public text added by Barth, Andreas/ARM:1187425 at 2016-11-22 16:32:38
Hello Joachim,
so, do I understand correct, you basically want to set the size for the Global Memory Pool:

http://www.keil.com/pack/doc/CMSIS/RTOS2/html/MemoryAllocation.html#GlobalMemoryPool

to 0.

In my simple example, that does not use user timers, it worked, after I also added custom control block memory to the attributes:

#include "cmsis_os2.h"                  // ARM::CMSIS:RTOS2:Keil RTX5
#include "rtx_os.h"                     // ARM::CMSIS:RTOS2:Keil RTX5

uint32_t Thread_1_Stack[ 0x1000 / sizeof( uint32_t ) ] ;
uint8_t Thread_1_TCB[ os_ThreadCbSize ] ;

  ThreadAttr.stack_mem = Thread_1_Stack ;
  ThreadAttr.stack_size = sizeof( Thread_1_Stack ) ;
  ThreadAttr.cb_mem = Thread_1_TCB ;
  ThreadAttr.cb_size = sizeof( Thread_1_TCB ) ;
  osThreadNew( Thread_1, NULL, &ThreadAttr ) ;    // Create application main thread

Does this help?
Best regards,
Andreas Barth
Public text added by Beck, Joachim/Unknown Company:1366868 at 2016-11-22 19:24:11
Hello Andreas,

no, that is not my problem.

I have the global memory set to 0, no problem. I use object specific memory for all RTOS objects. 
Try the following:
- set the thread memory model to object specific
- set the thread number to one
- set the default stack size to minimum (96), zero is not possible
- implement this thread with stack provided via attributes, as you have done in your example
- compile and look at the RW-data size
- now raise the default stack size to say 200
- compile and look at the RW-data size, it has raised. WHY? There is no thread which needs a default stack. 

This gets even worse if you allow more than one thread. I find no way to supress this unnecessary reservation of memory. 
I hope the problem is now clear.
 
Thank you for your support.

Best regards
Joachim Beck

------------------------------------------------------------------------------------------------------
Public text added by Barth, Andreas/ARM:1187425 at 2016-11-23 10:43:41
Ok, I see.
Theoretically I would say, in that case you also set "Number of user Threads with user-provided Stack size" to 1. However, this leads to the error:

#if ((OS_THREAD_USER_STACK_NUM != 0) && (OS_THREAD_USER_STACK_SIZE == 0))
#error "Total Stack size must not be zero!"
#endif

which probably dates from the time, when there was no way to provide thread stack really manually by the application. So this check could be removed (remove write protection and try it).

Then no stack for user threads should be allocated by the RTX5. Is this what you want to achieve?

Best regards,
Andreas Barth

------------------------------------------------------------------------------------------------------
Public text added by Beck, Joachim/Unknown Company:1366868 at 2016-11-23 11:04:23
Exactly, that is what I want to achieve.
I tried the same what you tried myself and got the same result. 
It seems to be a bug in the RTOS configuration. 
Could you give this information to the development team or should I open an issue on https://github.com/ARM-software/CMSIS_5?

Best regards
Joachim Beck

------------------------------------------------------------------------------------------------------
Public text added by Barth, Andreas/ARM:1187425 at 2016-11-23 11:41:21
Hello Joachim,
I am glad, we came to the point. Yes, I would suggest, you raise this on Github. Then other users are also immediately aware. Thank you.
Best regards,
Andreas Barth

ilg-ul, 2016-11-23:
Is this a CMSIS RTOS API issue, or a specific RTX issue?

joabeck, 2016-11-23:
I am not sure, but the error seems to be in line 71 of rtxConfig.h.
Regards
Joachim

RobertRostohar, 2016-11-24:
This is specific to RTX configuration and will be reviewed.

RobertRostohar, 2016-11-25:
RTX configuration has been updated to allow specifying the number of threads using default thread stack. Now it should be possible to avoid the unwanted stack memory reservation.

joabeck, 2016-11-28:
Ok, it is working now.
Thank you

closed 2016-11-28 by joabeck