ARM-software/CMSIS_5 #120: SVD Documentation: `addressBlock` element

ilg-ul, 2016-12-04:
The documentation page for the `addressBlock` element (http://www.keil.com/pack/doc/CMSIS/SVD/html/elem_peripherals.html#elem_addressBlock) 
lists `addressBlock` as child of `addressBlock`, while the schema does not.

Suggestion:

- update the documentation and remove the recursive definition

ilg-ul, 2016-12-08:
The documentation for register/fields (http://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_fields) reads `1..1` occurrences for `<field>`; probably it should read `1..*`.

ilg-ul, 2016-12-08:
The documentation for fields/field (http://www.keil.com/pack/doc/CMSIS/SVD/html/elem_registers.html#elem_field) reads `0..2` occurrences for `<enumeratedValues>`; probably it should read `0..*`,

jkrech, 2016-12-09:
The `<enumeratedValues>` tag is intentionally limited to a maximum of two instances.
This has to do with the tag `<usage>...</usage>`.  The usage can be either 
- read
- write
- read-write
You either have a single description used for read and write or two individual descriptions with the usage read and the other with the usage write.

bn01Mra, 2016-12-13:
- removed recursive element 'addressBlock' from 'addressBlock'.
- set occurrence of element 'field' in element 'fields' to [1..*]

I"ll update the documentation issues and submit the pages at the end of the day. Please wait until then.

closed 2016-12-13 by bn01Mra