ARM-software/CMSIS_5 #121: SVD Validator: two lines description elements

ilg-ul, 2016-12-05:
While using the SVD files, I noticed that sometimes the `<description>` text is arbitrarily split on two lines.

For example the following lines from the STM32F40x.svd:

```
            <field>
              <name>SEIS</name>
              <description>Seed error interrupt
              status</description>
              <bitOffset>6</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>CEIS</name>
              <description>Clock error interrupt
              status</description>
              <bitOffset>5</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
```

When parsing such a syntax, the result includes explicit line terminations and spaces:

```
	{
		"name": "SEIS",
		"description": "Seed error interrupt\r\n              status",
		"bitOffset": "6",
		"bitWidth": "1",
		"access": "read-write"
	},
	{
		"name": "CEIS",
		"description": "Clock error interrupt\r\n              status",
		"bitOffset": "5",
		"bitWidth": "1",
		"access": "read-write"
	},
```

This syntax leads to unusual displays in a debugger window. 

Identifying and fixing these strings at application level is possible, but requires some efforts.

Questions:

- Any idea why these strings got generated in this way?
- Would it be possible for the SVD validator to warn about these issues, to help those responsible for creating the SVD files identify and fix them? (the rule is: description strings split on two lines, without punctuation marks at the end of the first line)

JonatanAntoni, 2017-03-07:
The SVD files and description strings are handwritten or generating using vendor tool exports.
Unfortunately we have a lot of SVD files around without a common usage of the description field.

Instead of adding a bunch of messages (many SVDs are leading to thousands of them) we decided to overrule the formatting within out generators. Thus SVDConv for instance simply removes multiple white-spaces. Forcing a line-break can be achieved using a escaped linefeed (\n).

I have appended a description of the current usage to the [CMSIS-SVD documentation](https://www.keil.com/pack/doc/CMSIS/SVD/html/svd_Format_pg.html), see commit 5a5e78e08dc55d0c9497a0e0fca352472d692655.

JonatanAntoni, 2017-03-27:
Closed due to inactivity.

closed 2017-03-27 by JonatanAntoni