ARM-software/CMSIS_5 #122: [RTOS2] noTimeout (0ms) results in osErrorResource

DanielBoehmer, 2016-12-06:
Hi

The semaphore function `osSemaphoreAcquire `returns `osErrorResource `when 0 ms is used as timeout and no token is available to aquire.

I think `osErrorTimeout `would make more sense in this case since no token can be aquired in the given time and the application does not have to check multiple different return values if the timeout is a calcluated value between 0 and n milliseconds.

This problem is also available in `osThreadFlagsWait`.

Best regards,
Daniel

ilg-ul, 2016-12-06:
See #123 too.

JonatanAntoni, 2017-03-24:
@DanielBoehmer please take note about the updated documentation, i.e. [Semaphores](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__SemaphoreMgmt.html). To keep the interface simple we use a zero timeout to implement a `try_acquire`. Thus we return `osErrorResource` which might be read as a `would block`.
Please be so kind and close the issue if your concerns are resolved.

JonatanAntoni, 2017-04-03:
Closing this issue due to no further comments on the provided solution.

closed 2017-04-03 by JonatanAntoni