ARM-software/CMSIS_5 #123: CMSIS RTOSv2 API: return values for waiting functions

ilg-ul, 2016-12-06:
The documentation for `osSemaphoreAcquire()` is misleading. It not only refers to 'The _millisec_ timeout' while the parameter is in timer ticks, not in milliseconds, but the function is defined to return a `osStatus_t` value, and at the end it is mentioned: 'The return value indicates the number of available tokens (the semaphore count value). If 0 is returned, then no semaphore was available.'

No, the function should not return the number of available tokens, but a status code.

The problem is aggravated by the fact that this function multiplexes three different functionalities (since you refused to follow the advice to take the POSIX path and separate them, see #73).

The three separate cases have separate return values. Using the POSIX naming convention and the POSIX error codes, in the [CMSIS++](http://micro-os-plus.github.io/reference/cmsis-plus/group__cmsis-plus-rtos.html) implementation things look like this:

- `semaphore::wait(void)`
    - `result::ok` - The calling process successfully performed the semaphore lock operation.
    - `EPERM` - Cannot be invoked from an Interrupt Service Routines.
    - (other errors, like `EINTR`, `ENOTRECOVERBLE`)
- `semaphore::timed_wait(clock::duration_t timeout)`
    - `result::ok` - The calling process successfully performed the semaphore lock operation.
    - `EPERM` - Cannot be invoked from an Interrupt Service Routines.
    - `EINVAL` - Invalid timeout 
    - `ETIMEDOUT` - The semaphore could not be locked before the specified timeout expire
    - (other errors, like `EINTR`, `ENOTRECOVERBLE`)
- `semaphore::try_wait(void)`
    - `result::ok` - The calling process successfully performed the semaphore lock operation.
    - `EWOULDBLOCK` - The semaphore was already locked
    - (other errors, like `ENOTRECOVERBLE`)

Maybe the first two are similar enough to have them grouped under a single call, but in this case it must be mentioned that in the `osWaitForever` case there is no `ETIMEDOUT` return.

But the last one has a totally different semantic, does not return `EPERM` or `ETIMEDOUT`, instead it should return `EWOULDBLOCK` if the semaphore is not available.

The same approach should be consistently used for all objects that wait for a resource (`osMutexAcquire()`, `osMessageQueueGet()`, `osEventFlagsWait()`, `osThreadFlagsWait()`, etc)

JonatanAntoni, 2017-03-24:
We updated this part of the documentation early this year. Please take a look to the [Semaphores](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__SemaphoreMgmt.html) documentation and let us know about your concerns. If the documentation is clear to you now please be so kind and close this issue, thank you.

JonatanAntoni, 2017-04-03:
Closing this issue due to no further comments on the provided solution.

closed 2017-04-03 by JonatanAntoni