ARM-software/CMSIS_5 #124: Event Viewer

jankap, 2016-12-08:
Hello,

first thank you very much for this open development of the new Interface, looks promising.

I understood, the Event Recorder shall be used to capture OS events (and much more) which is very cool, but are there any plans to integrate the Keil uVision's Event Viewer (it's like the logic analyzer showing thread switches) again like it was in RTX (not RTX5)? In the HAL_CM.C file some methods 
```
void dbg_task_notify (P_TCB p_tcb, BOOL create);
void dbg_task_switch (U32 task_id);
void dbg_init (void);
```
are defined which use the ITM_Port 31. 

I don't know if you have something to do with the KEIL uVision development, but maybe starting here to ask is the right approach ;-)

Thank you.

belucha, 2016-12-17:
I also think, that the new API seems to be designed much more flexible. Congratulations!

Regarding the uVisions Event Viewer. I also miss the nice graphical view of thread switch events, however 
I saw support for for the Event Recorder was added in develop branch. I like this approach very much. Maybe it would be better to integrate a graphical representation of the recorder events to uVision, instead of supporting the legacy `dbg_task_notify` interface.

eugene345, 2017-01-20:
I have an experience of using both tools: Event Viewer and Event Recorder. I can say that Event Recorder is much more informative and better for debugging, finding bugs. But for time analysis, for sure, I prefer the Event Viewer with its graphical representation.  I hope that some kind of graphical tool will be added in future!

jankap, 2017-01-20:
Fyi, here http://www.keil.com/forum/62066/ somebody writes it's a known Bug and the Keil engineers would work on that...

Jonny-vb, 2017-03-30:
Is there any update on this? We were going to move over to RTOS2 until we discovered this was missing!

jankap, 2017-05-22:
Unfortunately I have to push this thread, too. Next gen product development will start soon, we have to decide if we can use CMSIS 5 or not.

jkrech, 2017-05-22:
As the visualization of the RTOS events is a product feature of MDK-ARM it is probably best discussed via a support ticket for MDK-ARM. Have you created a ticket on silver.arm.com already ?

Thanks
Joachim

Jonny-vb, 2017-05-23:
Surely it's not purely up to MDK-ARM though, and there must be something in the code that notifies the IDE of the thread switches?

JonatanAntoni, 2017-05-24:
Hi @Jonny-vb,

May I ask you to give me slightly more details about what you are doing, expecting and what is missing?

Did you notice the section [Add Event Recorder Visibility](http://www.keil.com/pack/doc/CMSIS/RTOS2/html/cre_rtx_proj.html#cre_rtx_proj_er) in the RTX5 documentation? If you set up EventRecorder with RTX5 you should already get a bunch of events like thread switches.

Regards,
Jonatan

Jonny-vb, 2017-05-24:
Hi @JonatanAntoni, Thanks for replying.

Using the CMSIS-RTOS1 with uvision, I am able to go to Debug->OS Support->Event Viewer and get a visualisation of which threads are running when. When I tried to do the same thing with a small test project using CMSIS-RTOS2, nothing was displayed. As others have mentioned, this is a very useful tool to get an overview of what is happening with the running threads, as well as timing information between them.

From what I can tell, the event recorder can provide this information, but does not display it in the graphical way that we currently use, unless I have missed something in it?

JonatanAntoni, 2017-05-24:
Hi @Jonny-vb, thank you for clarifying.

The Event Viewer is part of the deprecated CMSIS-RTOS1 awareness in MDK.  I am sorry for this inconvenience.

For CMSIS-RTOS2 you might use Event Recorder to get the events (only text based) and the new Component Viewer (View -> Watch Windows -> RTX RTOS). To get a graphical chart you might want to take a look to [Tracealyzer](https://www2.keil.com/mdk5/debug/tracealyzer/).

These all are tool related issues not CMSIS ones. So please consider to file a [customer request](http://silver.arm.com/) to raise attention on the tool side.

Jonny-vb, 2017-05-24:
Thanks for clarifying @JonatanAntoni, I was not aware the event viewer had been depreciated. It's a shame as having this built in to MDK and being able to see it in real time was a great feature. Tracealyzer does look like a good alternative though, although from what I can tell it doesn't have the same real time capabilities, and it's more money to spend!

I guess this issue is closed then (although it's not my issue to close!) Thanks again for the help.

closed 2017-05-24 by JonatanAntoni

JonatanAntoni, 2017-06-08:
Hi @Jonny-vb,

I got the information from MDK team that they are currently working on a successor of Event Viewer. According to the current release plan the new feature should become public available until the end of the year.

jankap, 2017-06-08:
@JonatanAntoni 
Thanks for giving us this information. Do you know if this is based on Tracealyzer? We're currently evaluating Tracealyzer and consider buying it, but if it's going to be part of uVision this would be completely useless.

JonatanAntoni, 2017-06-08:
@jankap please get in touch with MDK customer support for tool related questions.

jankap, 2017-06-12:
Just FYI, 
I asked the ARM Support this question:

"Is there a development of a "new" Event Viewer, or any graphical tool to display the Event Recorder's messages (I like the idea of Event Recorder, but without visualization it's not as useful as it could be)?
Since Keil/ARM and Percepio seem to work closer together, is there any chance that some kind of tracealyzer will be integrated into uVision? If yes - in which version will it be available and in what timeframe?"

Answer:
"Work is ongoing to have a replacement for the Event Viewer functionality. But right now not sure when it will become available. What breaks the old Event Viewer feature is, that legacy RTX used 16 bit TaskID values. Recent CMSIS RTOS now us the 32bit TCB pointer as TreadID. So there will be a new solution based on the Event Recorder."