ARM-software/CMSIS_5 #125: Mutex documentation is misleading

bulislaw, 2016-12-08:
Documentation says `The advantage of a mutex is that it introduces thread ownership. When a thread acquires a mutex and becomes its owner, subsequent mutex acquires from that thread will succeed immediately without any latency. Thus, mutex acquires/releases can be nested.`

But acquiring mutex from a thread twice results in a hang (or timeout). To actually enable what the docs are saying you need to set `.attr_bits = osMutexRecursive`, the high level documentation should mention that.

JonatanAntoni, 2017-03-13:
The documentation was updated in the meanwhile. It states 
> When a thread acquires a mutex and becomes its owner, subsequent mutex acquires from that thread will succeed immediately without any latency (if osMutexRecursive is specified). Thus, mutex acquires/releases can be nested.
Is this clear enough now? Please close this issue if you do not have any further questions.

bulislaw, 2017-03-13:
Thanks, I think it's clear now. Closing.

closed 2017-03-13 by bulislaw